freebsd/crochet #101: Selecting Configuration File for UBoot

rnikoopour, 2015-01-30:
Hello,

I'm trying to build an image for the Raspberry Pi and I've run into some trouble when Crochet builds uboot.

I have changed settings in < u-boot-rpi/include/configs/rpi_b.h > to disable to serial console and make the autoboot time 10 seconds instead of 3.  I deleted < crochet-freebsd/work > directory to build from scratch.

I wrote the image created using dd.  When I booted my RaspberryPi none of the changes I made to the rpi_b.h were there.

How do I get Crochet to build uboot using the settings I defined in < crochet-freebsd/u-boot-rpi/include/configs/rpi_b.h >?

Thanks,
Reza

kientzle, 2015-01-31:
> On Jan 30, 2015, at 3:41 PM, rnikoopour <notifications@github.com> wrote:
> 
> Hello,
> 
> I'm trying to build an image for the Raspberry Pi and I've run into some trouble when Crochet builds uboot.
> 
> I have changed settings in < u-boot-rpi/include/configs/rpi_b.h > to disable to serial console and make the autoboot time 10 seconds instead of 3. I deleted < crochet-freebsd/work > directory to build from scratch.
> 
> I wrote the image created using dd. When I booted my RaspberryPi none of the changes I made to the rpi_b.h were there.
> 
> How do I get Crochet to build uboot using the settings I defined in < crochet-freebsd/u-boot-rpi/include/configs/rpi_b.h >?
> 
> 

Crochet is probably just re-using the previously
built version.

To get Crochet to rebuild U-Boot, you should look in
the crochet-freebsd/u-boot-rpi directory and remove *only*
these two files:
   _.uboot.configured
   _.uboot.built

If these files are present, Crochet will skip the configuration
and building of U-Boot.  Removing them will allow Crochet to
configure U-Boot again and then rebuild it all.

Not: you probably should *not* remove the
  _.uboot.patched
file; if you remove that, Crochet will try to re-apply its
patches which will probably fail.  If that happens, you’ll
need to start from a clean checkout of the U-Boot
sources.

Good luck!

Tim

rnikoopour, 2015-01-31:
Hey Tim!

Thank you for your help!  I figured it shortly before your post =P

Have a good one!

closed 2015-01-31 by rnikoopour