freebsd/crochet #111: Some options not being adhered in the finished image.

dinapappor, 2015-05-18:
When I build an image with the crochet tool for rpi2 I have two options that doesn't seem to be honoured.

Option Growfs

The / is not resized to use maximum amount of the card (this is a 8 GB card), I am only getting 
/dev/mmcsd0s2a    3.5G    1.8G    1.3G    58%    /

When running df -h

option SwapFile 512mb file=/usr/swap0
Swapfile is created but is not used. There is no output in top for it either.

dinapappor, 2015-05-18:
Adding late option to swap-file option got the swap mounted for me.
md none swap sw,file=/usr/swap0,late 0 0

kientzle, 2015-05-22:
I've added 'late' to fstab entry created for SwapFile
Thank you for pointing that out.

I'll try using the Growfs option on my own next image and see if I can puzzle anything out.

kientzle, 2015-05-25:
Ah.  I believe this is all fixed now.  The key issue for growfs:  Crochet is now using the standard /etc/rc.d/growfs script to handle this, which in turn relies on "firstboot" logic which requires a new image to have a file called /firstboot 

Crochet now adds a /firstboot file when installing world.

closed 2015-05-25 by kientzle