freebsd/crochet #113: need cpp stage of compilation for buil after r283276

f-andrey, 2015-05-24:
After https://svnweb.freebsd.org/base?view=revision&revision=283276 there may be problems with the build of some of the platforms that I encountered during build BEAGLEBONE.

Info for compilation error and recommendations @gonzoua
https://lists.freebsd.org/pipermail/freebsd-arm/2015-May/011486.html

kientzle, 2015-05-24:
Copying Gonzo's comments here for reference:

> #include in dts file is handled by cpp(1). /include/ is handled
> by dtc I believe
>
> You can take a look at how FreeBSD compiles dts files in
> sys/tools/fdt/make_dtb.sh
>
> crochet does not have cpp stage of compilation and before my TI
> code/devicetree refactoring none of the dts files referenced in
> crochet used #include. That's why problem never appeared. 
>
> Fix is just a matter of fixing freebsd_install_fdt in lib/freebsd.sh.
> If nobody beats me to it I'll try to fix it and submit pull request to Tim.

kientzle, 2015-05-25:
Should be fixed now.

closed 2015-05-25 by kientzle