freebsd/crochet #116: WARNING: pw after r284118 not execute -V

f-andrey, 2015-06-14:
Hello,
```shell
/usr/sbin/pw useradd -n user -s /bin/csh -g wheel -w yes -V /usr/obj/_.mount.freebsd/etc/ -d /usr/home/user
```
make not user with /usr/obj/_.mount.freebsd/etc/ but creates it in the
main system /etc

mail list message https://lists.freebsd.org/pipermail/freebsd-current/2015-June/056208.html

bapt, 2015-06-14:
According to the manpage the -V switch was always supposed to be before the "useradd" subcommand, the lastest modifications enforced that by accident. I can be convinced to make it accepted like used in crochet, but IMHO having the code respecting the documentation is better (plus within the code it makes more sense :D.

kientzle, 2015-06-14:
> On Jun 14, 2015, at 2:02 PM, Baptiste Daroussin <notifications@github.com> wrote:
> 
> According to the manpage the -V switch was always supposed to be before the "useradd" subcommand, the lastest modifications enforced that by accident. I can be convinced to make it accepted like used in crochet, but IMHO having the code respecting the documentation is better (plus within the code it makes more sense :D
> 
> 

It probably does make more sense and I’ll definitely update Crochet’s usage.  To make the documentation a little clearer, please change “may” to “must” in the man page:

"As an exception to the general rule where options must follow the operation type, the -V flag may be used on the command line before the operation keyword.”

Cheers,

Tim

bapt, 2015-06-14:
Done, please note that a new -r <rootdir> has been added, which could be useful for crochet (will be MFC next week)

kientzle, 2015-06-15:
I've changed Crochet to put the -V option before the subcommand.

Please update and try again.

closed 2015-06-15 by kientzle