freebsd/crochet #118: phase 510 fails: Couldn't open output file ... Is a directory

dgrechka, 2015-06-29:
System: FreeBSD 10.1-RELEASE-p10 FreeBSD 10.1-RELEASE-p10 #8 r283762
u-boot-rpi revision: 6709570cd

Building the standard FreeBSD image fails on phase 510:

root@cube:/tmp/crochet # ./crochet.sh -b RaspberryPi -v -v -v
Starting at Mon Jun 29 10:53:25 MSK 2015
Board: RaspberryPi
Source version is: 284881
Building FreeBSD version: 10.1
====================> Phase 100 PHASE_POST_CONFIG <====================
Running:  OPTION= OPTIONDIR= BOARDDIR= board_mountpoint_defaults
Running:  OPTION= OPTIONDIR= BOARDDIR= board_defined
Running:  OPTION= OPTIONDIR= BOARDDIR= install_customize_hooks
Running:  OPTION= OPTIONDIR= BOARDDIR= freebsd_default_makeobjdirprefix
Running:  OPTION= OPTIONDIR= BOARDDIR= board_generate_image_name
Image name is:
    /tmp/crochet/work/FreeBSD-armv6-10.1-RPI-B-284881.img
====================> Phase 110 PHASE_CHECK <====================
Running:  OPTION= OPTIONDIR= BOARDDIR= freebsd_current_test
Building FreeBSD version: 10.1
Object files are at: /tmp/crochet/work/obj/arm.armv6/usr/src
Found suitable FreeBSD source tree in:
    /usr/src
Running:  OPTION= OPTIONDIR= BOARDDIR= board_check_image_size_set
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi mkimage_python_check
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi raspberry_pi_check_uboot
Found FreeBSD xdev tools for armv6
Found U-Boot sources in:
    /tmp/crochet/u-boot-rpi
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi raspberry_pi_check_bootcode
====================> Phase 210 PHASE_BUILD_WORLD <====================
Running:  OPTION= OPTIONDIR= BOARDDIR= board_default_buildworld
Building FreeBSD armv6 world at Mon Jun 29 10:53:27 MSK 2015
    (Logging to /tmp/crochet/work/_.buildworld.armv6.log)
====================> Phase 220 PHASE_BUILD_KERNEL <====================
Running:  OPTION= OPTIONDIR= BOARDDIR= board_default_buildkernel
Building FreeBSD armv6-RPI-B kernel at Mon Jun 29 13:11:39 MSK 2015
    (Logging to /tmp/crochet/work/_.buildkernel.armv6-RPI-B.log)
====================> Phase 230 PHASE_BUILD_OTHER <====================
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi freebsd_ubldr_build UBLDR_LOADADDR=0x2000000
Building FreeBSD armv6-RPI-B ubldr at Mon Jun 29 13:24:32 MSK 2015
    (Logging to /tmp/crochet/work/ubldr-armv6-RPI-B/_.ubldr.armv6-RPI-B.build.log)
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi uboot_patch /tmp/crochet/u-boot-rpi /tmp/crochet/board/RaspberryPi/files/uboot-rpi_patch10_makefile_lc.patch /tmp/crochet/board/RaspberryPi/files/uboot-rpi_patch11_examples_api_makefile_static.patch /tmp/crochet/board/RaspberryPi/files/uboot-rpi_patch13_sdhci_no_high_speed.patch
Patching U-Boot at Mon Jun 29 13:24:38 MSK 2015
    (Logging to /tmp/crochet/u-boot-rpi/_.uboot.patch.log)
/tmp/crochet/board/RaspberryPi/files/uboot-rpi_patch10_makefile_lc.patch
/tmp/crochet/board/RaspberryPi/files/uboot-rpi_patch11_examples_api_makefile_static.patch
/tmp/crochet/board/RaspberryPi/files/uboot-rpi_patch13_sdhci_no_high_speed.patch
   Applying patch /tmp/crochet/board/RaspberryPi/files/uboot-rpi_patch10_makefile_lc.patch
   Applying patch /tmp/crochet/board/RaspberryPi/files/uboot-rpi_patch11_examples_api_makefile_static.patch
   Applying patch /tmp/crochet/board/RaspberryPi/files/uboot-rpi_patch13_sdhci_no_high_speed.patch
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi uboot_configure /tmp/crochet/u-boot-rpi rpi_b_config
Configuring U-Boot at Mon Jun 29 13:24:38 MSK 2015
    (Logging to /tmp/crochet/u-boot-rpi/_.uboot.configure.log)
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi uboot_build /tmp/crochet/u-boot-rpi
Building U-Boot at Mon Jun 29 13:24:39 MSK 2015
    (Logging to /tmp/crochet/u-boot-rpi/_.uboot.build.log)
====================> Phase 301 PHASE_IMAGE_BUILD_LWW <====================
Running:  OPTION= OPTIONDIR= BOARDDIR= board_default_create_image
Creating a 1000MB raw disk image in:
    /tmp/crochet/work/FreeBSD-armv6-10.1-RPI-B-284881.img
====================> Phase 311 PHASE_PARTITION_LWW <====================
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi raspberry_pi_partition_image
Partitioning the raw disk image with MBR at Mon Jun 29 13:25:11 MSK 2015
gpart create -s MBR md4
md4 created
Creating a 17m FAT partition at Mon Jun 29 13:25:11 MSK 2015 with start block 63 and label BOOT
active set on md4s1
Creating an auto-sized UFS partition at Mon Jun 29 13:25:11 MSK 2015
md4s2 created
/dev/md4s2a: 936.6MB (1918080 sectors) block size 32768, fragment size 4096
        using 4 cylinder groups of 234.16MB, 7493 blks, 30080 inodes.
super-block backups (for fsck_ffs -b #) at:
 192, 479744, 959296, 1438848
tunefs: soft updates set
Using inode 4 in cg 0 for 4194304 byte journal
tunefs: soft updates journaling set
tunefs: NFSv4 ACLs set
====================> Phase 321 PHASE_MOUNT_LWW <====================
Running:  OPTION= OPTIONDIR= BOARDDIR= board_default_mount_partitions
Mounting all file systems:
Mounting FAT partition 1 at /tmp/crochet/work/_.mount.boot
Mounting UFS partition 1 at /tmp/crochet/work/_.mount.freebsd
====================> Phase 510 PHASE_BOOT_INSTALL <====================
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi raspberry_pi_populate_boot_partition
FATAL ERROR: Couldn't open output file /tmp/crochet/work/fdt/fdt.HMeUPW: Is a directory

cp: /tmp/crochet/work/fdt/fdt.HMeUPW/*.dtb: No such file or directory

kientzle, 2015-06-30:
Can you use FreeBSD 10-STABLE instead?

There is a known problem with FreeBSD 10.1 (the DTB handling in FreeBSD changed just after FreeBSD 10.1 was released).

Tim

dgrechka, 2015-06-30:
Hi,

but when I take older crochet version (e.g. fb14276) and build the same image on the same system the re is no such issue:

====================> Phase 510 PHASE_BOOT_INSTALL <====================
Running:  OPTION= OPTIONDIR= BOARDDIR=/tmp/crochet/board/RaspberryPi raspberry_pi_populate_boot_partition
python imagetool-uncompressed.py /tmp/crochet/u-boot-rpi/u-boot.bin - > uboot.img
Installing ubldr in .
====================> Phase 711 PHASE_FREEBSD_INSTALLWORLD_LWW <====================

As the present version uses DTB (I guess rev.fb14276 does not) maybe Crochet should stop at the very begining explaining the problem and proposing the ways to solve it?

praetor242, 2015-07-16:
I too am having the exact same problem. Any idea on a fix or workaround?

dgrechka, 2015-07-16:
I've checked out older code revision fb14276 and it built successfully.

kientzle, 2015-07-17:
The easiest workaround is to get FreeBSD 10-STABLE source instead of FreeBSD 10.1.  Or any version later than 10.1.

IF you must build FreeBSD 10.1, you can workaround by editing one line in lib/freebsd.sh.  That line 436 reads:

echo ${FREEBSD_SRC}/sys/tools/fdt/make_dtb.sh ${FREEBSD_SRC}/sys ${_DTSIN} ${_DTBINTERMEDIATEDIR} | (cd ${FREEBSD_SRC}; make TARGET_ARCH=$TARGET_ARCH buildenv > /dev/null)

Change it by adding "/out.dtb" after ${_DTBINTERMEDIATEDIR} so it looks like this:

echo ${FREEBSD_SRC}/sys/tools/fdt/make_dtb.sh ${FREEBSD_SRC}/sys ${_DTSIN} ${_DTBINTERMEDIATEDIR}/out.dtb | (cd ${FREEBSD_SRC}; make TARGET_ARCH=$TARGET_ARCH buildenv > /dev/null)

Again, this is ONLY for FreeBSD 10.1.  Later versions of FreeBSD (including 10-STABLE and the upcoming 10.2) require a directory name at that point.

I'm trying to come up with a way for Crochet to detect which version of FreeBSD you're trying to build and do the right thing in both cases.

kientzle, 2015-07-18:
I just committed an update that checks the version of FreeBSD you're building and invokes the DTB logic appropriately.  This should allow Crochet to build FreeBSD 10.1 or later versions without issues.

closed 2015-07-18 by kientzle