freebsd/crochet #126: _freebsd_install_usr_src() should ignore SCM data

scsiguy, 2015-09-04:
It's pretty easy to overflow the memory disk used for prepping the image if you enable copying of /usr/src and it happens to have been fetched via git. Adding '--exclude .git' to the tar command worked for me, but the function should probably ignore other types of SCM data too.

kientzle, 2015-09-04:
This is a little tricky:  Personally, when I include /usr/src on an image, I *want* to be able to update it using the standard tools.  (Especially since doing a straight SVN checkout directly on RPi or Beaglebone is so very, very, very painful.  I would really appreciate if more people tried this and helped come up with ideas for how to fix the obvious problems.)

I suppose the only fix is probably to add an argument to control this.  Then we argue about which way should be default.  ;-)

brd, 2015-09-04:
I'll volunteer for doing the work.. It might be better to default to excluding the .$vcs dir..