freebsd/crochet #127: Patch src before build?

brd, 2015-09-11:
Since the BeagleBone Black having support for ethernet/serial over the usb cable, I'd like crochet to patch the src tree to enable this support. It is really a small patch:

```
Index: usb/usb_device.c
===================================================================
--- usb/usb_device.c    (revision 287099)
+++ usb/usb_device.c    (working copy)
@@ -112,6 +112,8 @@

 /* This variable is global to allow easy access to it: */

+#define        USB_TEMPLATE USB_TEMP_SERIALNET
+
 #ifdef USB_TEMPLATE
 int    usb_template = USB_TEMPLATE;                                                                          
 #else
```

Any thoughts on implementing something like this?

My concern is if someone is building from /usr/src we don't want to go changing that on them.. but we could detect if the source tree is /usr/src and revert the patch after the build?

kientzle, 2015-09-12:
Crochet should never patch the FreeBSD source tree.

You should send your suggested change to the FreeBSD developers and see about getting it adopted there.  (It may just involve a simple change to the BeagleBone kernel configuration.)

closed 2015-09-12 by kientzle

brd, 2015-09-12:
Currently it cannot default to on in the main tree. What about having Crochet manage its own tree?

erikarn, 2015-09-12:
Create a patch for freebsd-head that adds that as a kernel-time build option. It should be pretty trivial to do. Like, it should one line in an options file, and one line in a kernel config file.

Done aaaaand done.

erikarn, 2015-09-12:
.. also, since it's assigning some value to an int, you can also likely make it a boot time tunable, which also means it can just be customised into /boot/loader.conf - that's something crochet can also do.

If you'd like some help on doing this then please let me know and grab me on irc. :)