freebsd/crochet #128: Rolling ports and special configurations into build image

soylentgreen, 2015-09-20:
Hi, 

This isn't so much of an issue, but I'm more curious whether it's possible or not. 

I'd like to build the Raspberry Pi 2 images with X11 and also have my own configuration files which would override original configuration files. 

So, the questions I have are: 

1.) Is there a way to add Xorg (possibly Wayland) during the build process and not after? 
2.) Is there a way to specify my own configuration files during the build process which would override the resulting configuration files from the installation of other programs? IE: If I create an Xorg.conf and I want to use the settings in that vs what's generated by Xorg, would I be able to slipstream those files into the installation?

kientzle, 2015-09-20:
> On Sep 20, 2015, at 8:58 AM, soylentgreen <notifications@github.com> wrote:
> 
> Hi,
> 
> This isn't so much of an issue, but I'm more curious whether it's possible or not.
> 
> I'd like to build the Raspberry Pi 2 images with X11 and also have my own configuration files which would override original configuration files.
> 
> So, the questions I have are:
> 
> 1.) Is there a way to add Xorg (possibly Wayland) during the build process and not after? 

Sort of.  There's been some attempt to support package installation during the build process (see 'option Package' for the current attempt), but it's still a little rough:
  * You have to have the packages, of course.  This may require bringing up an RPI2, building the software there, then packaging it and copying the packages to your build machine.
  * The 'pkg' command only handles cross-installs for certain packages.  ("Cross-install" meaning that the actual installation onto the image occurs on a non-ARM system.)  The problem is that many packages need to run software on the target as part of the installation.  The 'pkg' developers have been working to reduce this requirement but I don't know how far they've managed to push it.

An alternative would be to include the package on the image and do the installation per se on first boot.  I've not explored that approach but it should be straightforward.

> 2.) Is there a way to specify my own configuration files during the build process which would override the resulting configuration files from the installation of other programs? IE: If I create an [...]

Yes.  Crochet has very robust support for overwriting configuration at the end of building the image.

Tim

soylentgreen, 2015-09-21:
Hi Tim! 

Thanks for your reply. Very helpful!

In response to your Xorg response... What if I instead specified to cross compile Xorg to ARM during the build process instead of using a package? Would this have the same limitation?

kientzle, 2015-09-21:
>  What if I instead specified to cross compile Xorg to ARM during the build process instead of using a package?

If you figure out how to do this, please let me know.  It's an interesting idea.

closed 2015-09-21 by kientzle

brd, 2015-09-21:
I am working on patches to the ports framework to support this. The next version of pkg will have better support for this and then the ports framework will need a bunch of TLC to respect what pkg is doing. I hope to have that committed this week and get another package run done on the build cluster.