freebsd/crochet #136: add start_x.elf to Raspberry PI boot

serge-v, 2015-11-13:
Need to add 2 new RPI firmware files start_x.elf and fixup_x.dat.
They are required when start_x=1 config.txt option is specified.

brd, 2015-11-13:
Can you explain a bit more about why they are required?

serge-v, 2015-11-13:
Camera didn't work with start_x=0 in config.txt.
Raspistill is complaining that mmal is not enabled.

I tried enable the camera using these instructions:
http://raspberrypi.stackexchange.com/questions/14229/how-can-i-enable-the-camera-without-using-raspi-config
and
http://blog.mcmelectronics.com/post/Getting-Up-and-Running-with-the-Raspberry-Pi-Camera-Board#.VkXyI66rTOQ

So I set:
start_x=1
PRI doesn't boot at all.
After copying start_x.elf and fixup_x.dat and then it boots successfully
and camera is working.

kientzle, 2015-11-13:
Are those files in the RPi U-Boot port?   If not, we should get them added there.  I believe Crochet just copies all the files from that port.

serge-v, 2015-11-13:
u-boot-rpi/Makefile doesn't have any references to start_x.elf and fixup_x.dat. Is this correct place to check?

Makefile:
https://svnweb.freebsd.org/ports/head/sysutils/u-boot-rpi/Makefile?revision=400051&view=markup

brd, 2015-11-13:
I wrote up these instructions a few weeks ago to get my camera working:

https://wiki.freebsd.org/FreeBSD/arm/Raspberry%20Pi%20Camera

serge-v, 2015-11-13:
Do you have start_x.elf on RPI2 boot partition?

brd, 2015-11-13:
Yes.

kientzle, 2015-11-14:
> On Nov 13, 2015, at 6:14 AM, Serge Voilokov <notifications@github.com> wrote:
> 
> Need to add 2 new RPI firmware files start_x.elf and fixup_x.dat.
> They are required when start_x=1 config.txt option is specified.

To clarify, is this for RPI or RPI2?

It looks like the RPI2 U-Boot port does have those files but
they're missing from the RPI port.

In either case, Crochet should just be copying all the files from
the port, so if something is missing, it should be corrected in
the port.

serge-v, 2015-11-15:
I have RPI-A.
Right, sysutils/u-boot-rpi2 has start_x. u-boot-rpi has not. Will add. Thanks for clarification.

kientzle, 2016-11-22:
Closing this old issue, since it was really a problem with the RPI U-Boot port.

closed 2016-11-22 by kientzle