freebsd/crochet #137: Raspberry Pi 2: don't use md for /var/log

kgadek, 2015-12-09:
Looking at [most recent /etc/fstab for RPi2](https://github.com/freebsd/crochet/blob/1c44bd3b2e769fdbf223236d4e8901e967715660/board/RaspberryPi2/overlay/etc/fstab) and it's contents is:

```
/dev/mmcsd0s1	/boot/msdos	msdosfs rw,noatime	0 0
/dev/mmcsd0s2a	/		ufs rw,noatime		1 1
md		/tmp		mfs rw,noatime,-s50m	0 0
md		/var/log	mfs rw,noatime,-s15m	0 0
md		/var/tmp	mfs rw,noatime,-s10m	0 0
```

Notably, `/var/log` is memory based. IMHO, this is bad.

Is there a reason to do this? If not, we could probably just drop this line and allow `/var/log` to live under root.

brd, 2015-12-09:
Generally, logging on devices like these isn't all that useful and will chew up the cells on the flash device, so we avoid it by putting it on a memory device by default.

kgadek, 2015-12-10:
Ok, for me logs are never to be lost. However, if the community wants ephemeral logs, let it be.

closed 2015-12-10 by kgadek

brd, 2015-12-10:
You might consider setting syslog up to log over the network to a server.  That is what I do when I care about logs never being lost.

Thanks for the feedback though :)