freebsd/crochet #138: What I miss ? I cant build

giovannimanzoni, 2015-12-22:
Hello, I can not compile for any board.. 

I will get, for any board, that directory ddo not exist. : 

KERNEL=kernel all -DNO_MODULES_OBJ
cd: /usr/home/gio/crochet/work/obj/arm.armv6/usr/src/sys/XXXXXXXXX: No such file or directory
*** [buildkernel] Error code 2

My logs:

[gio@pcbsd-i7] ~/crochet% time sudo /bin/sh crochet.sh -c config.sh
Starting at Mar 22 Dic 2015 02:28:35 CET
Loading configuration from config.sh
Board: RaspberryPi2
Option: ImageSize 1950mb
Source version is: 292577
Building FreeBSD version: 11.0
Image name is:
    /usr/home/gio/crochet/work/FreeBSD-armv6-11.0-RPI2-292577.img
Building FreeBSD version: 11.0
Object files are at: /usr/home/gio/crochet/work/obj/arm.armv6/usr/src
Found suitable FreeBSD source tree in:
    /usr/src
Found U-Boot port in:
    /usr/local/share/u-boot/u-boot-rpi2
Building FreeBSD armv6 world at Mar 22 Dic 2015 02:28:52 CET
    (Logging to /usr/home/gio/crochet/work/_.buildworld.armv6.log)
Building FreeBSD armv6-RPI2 kernel at Mar 22 Dic 2015 03:01:51 CET
    (Logging to /usr/home/gio/crochet/work/_.buildkernel.armv6-RPI2.log)
Failed to build FreeBSD armv6-RPI2 kernel.
Log in /usr/home/gio/crochet/work/_.buildkernel.armv6-RPI2.log
                                                                                                                                                       
make[1]: stopped in /usr/src
1 error
                                                                                                                                                       
make[1]: stopped in /usr/src
*** [buildkernel] Error code 2
                                                                                                                                                       
make: stopped in /usr/src
1 error

make: stopped in /usr/src
18909.485u 1364.680s 33:18.28 1014.5% 43926+660k 210263+2004464io 40453pf+0w
[gio@pcbsd-i7] ~/crochet% 

cat /usr/home/gio/crochet/work/_.buildkernel.armv6-RPI2.log
--- buildkernel ---
--- buildkernel ---
--------------------------------------------------------------
>>> Kernel build for RPI2 started on Tue Dec 22 03:01:51 CET 2015
--------------------------------------------------------------
===> RPI2
mkdir -p /usr/home/gio/crochet/work/obj/arm.armv6/usr/src/sys
--------------------------------------------------------------
>>> stage 2.3: build tools
--------------------------------------------------------------
cd /usr/src; TOOLS_PREFIX=/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp MAKEOBJDIRPREFIX=/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp INSTALL="sh /usr/src/tools/install.sh" PATH=/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/legacy/usr/sbin:/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/legacy/usr/bin:/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/legacy/bin:/sbin:/bin:/usr/sbin:/usr/bin WORLDTMP=/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp make -f Makefile.inc1 DESTDIR= BOOTSTRAPPING=1002000 SSP_CFLAGS= MK_HTML=no -DNO_LINT MK_MAN=no -DNO_PIC MK_PROFILE=no -DNO_SHARED -DNO_CPU_CFLAGS MK_WARNS=no MK_CTF=no kernel-tools
--- kernel-tools ---
mkdir -p /usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/usr
mtree -deU -f /usr/src/etc/mtree/BSD.usr.dist -p /usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/usr >/dev/null
--------------------------------------------------------------
>>> stage 3.2: building everything
--------------------------------------------------------------
cd /usr/home/gio/crochet/work/obj/arm.armv6/usr/src/sys/RPI2; MAKEOBJDIRPREFIX=/usr/home/gio/crochet/work/obj/arm.armv6 MACHINE_ARCH=armv6 MACHINE=arm CPUTYPE= GROFF_BIN_PATH=/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/legacy/usr/bin GROFF_FONT_PATH=/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/legacy/usr/share/groff_font GROFF_TMAC_PATH=/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/legacy/usr/share/tmac CC="cc " CXX="c++ " DEPFLAGS="" CPP="cpp " AS="as" AR="ar" LD="ld" NM=nm OBJDUMP=objdump OBJCOPY="objcopy" RANLIB=ranlib STRINGS= SIZE="size" _LDSCRIPTROOT= INSTALL="sh /usr/src/tools/install.sh" PATH=/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/legacy/usr/sbin:/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/legacy/usr/bin:/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/legacy/bin:/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/usr/sbin:/usr/home/gio/crochet/work/obj/arm.armv6/usr/src/tmp/usr/bin:/sbin:/bin:/usr/sbin:/usr/bin make -D KERNFAST -j 24 -J 15,16 -m /usr/src/share/mk KERNEL=kernel all -DNO_MODULES_OBJ
cd: /usr/home/gio/crochet/work/obj/arm.armv6/usr/src/sys/RPI2: No such file or directory
*** [buildkernel] Error code 2

make[1]: stopped in /usr/src
1 error

make[1]: stopped in /usr/src
*** [buildkernel] Error code 2

make: stopped in /usr/src
1 error

make: stopped in /usr/src
[gio@pcbsd-i7] ~/crochet%

giovannimanzoni, 2015-12-22:
Hello, I find the problem, ona package missing in my pcbsd: devel/gmake. thanks to https://www.raspberrypi.org/forums/viewtopic.php?f=85&t=90613

brd, 2015-12-23:
That is strange. I'll dig in and see what would be depending on it and see if we can either remove it or add a check for it as needed.

kientzle, 2015-12-24:
We used to need gmake to build U-Boot, but now we're using the port, so I don't know that it is needed.

It's definitely not needed for building the FreeBSD kernel which is what Giovanni was having trouble with, so I suspect there is something else going on there.

giovannimanzoni, 2015-12-25:
Hello, yes, I'm in truble with an other things. I had intended to reopen the issue and now I see that you've been paying attention to my report. Wonderfull !!  I think first I was in truble with FreeBSD repositories, and Today I  'm not sure to use https://svn.freebsd.org/base/release/10.2.0/

I fork your repo to https://github.com/giovannimanzoni/crochet , for my test. I think I add some beautifull improvement, you can diff. My spirit is to have a script full working without prompt to user to do thing. I think that nowdays we can write automatic script to handle all exception in the best way. 

I force git  to add config.sh so you can test exactly my configuration. 

If you try my version of script, during compilation it fail with 

[fail build.txt](https://github.com/freebsd/crochet/files/71968/fail.build.txt)
and this is the log
[log.txt](https://github.com/freebsd/crochet/files/71969/log.txt)

I've 24 thread i7 cpu, for see that error, I've to stay on the monitor for see the message on the screen. Only now I see this error between compiler message...I don't know/remember how, but one time I was able to make full working image, flash and boot BBB with FreeBSD and in this time I do not know it the message has been shown and the compiler create the image anyway.

My final proposal is to add FOXG20 and than some other boards to your script like Pine64, and Zynq based boards. I've Raspberry and BBB so I can test how your script work.

I'm new to recompile FreeBSD and I would very very like to know what I miss. I've 10+ years experience in Linux embedded but I'm new to FreeBSD porting

Thanks !

giovannimanzoni, 2015-12-25:
@brd : "That is strange. I'll dig in and see what would be depending on it and see if we can either remove it or add a check for it as needed."

Yes you are right, I See this package was missing but then I follow some other howt-to's for rasberry and than back to BBB and than magically I build image for BBB. But when I try to clean and redo all think, same problem return.

giovannimanzoni, 2015-12-25:
Sorry, 12 thread so I can do make -j 24 and in an hour or minus I can repeat a clean rebuild

kientzle, 2015-12-25:
If you are having problems, you should re-run everything with *no* -j option.  The parallel builds mix a lot of messages which makes it hard to tell what command caused the actual error.

giovannimanzoni, 2015-12-25:
Ok I will run the script with no -j ... But what about repo to use ? And my fork ? But I read that rebuild freebsd is -j safe..

giovannimanzoni, 2015-12-27:
Now I've find raspbsd.org where there is bbb image made with FreeBSD 11 and specific svn revision. I Try this for see if make error will show up.

@brd: Whit which commamd you buil crosscompiling tools? With make XDEV=arm XDEV_ARCH=armv6 xdev ?

giovannimanzoni, 2015-12-27:
Some news...  with FreeBSD sourcer revision 291838

in my freebsd_src_build ( )  of my fork I try to do
-> https://lists.freebsd.org/pipermail/freebsd-arm/2015-July/011762.html
make -j 24 TARGET=arm TARGET_ARCH=armv6 buildworld
make -j 24 TARGET=arm TARGET_ARCH=armv6 KERNCONF=BEAGLEBONE buildkernel
-> http://www.onemansanthology.com/blog/freebsd-on-beaglebone-black/
make ${SRCJOBS} XDEV=arm XDEV_ARCH=armv6 xdev

building step go a bit next last point but fail
[log.txt](https://github.com/freebsd/crochet/files/72749/log.txt)

kientzle, 2015-12-28:
>> I read that rebuild freebsd is -j safe..

Generally, it is safe.  But when there are errors, -j makes the messages harder to read.

Also, the cross-compile tools are only needed for building U-Boot.  You should be using the U-Boot port instead.  Then you don't need the cross-compile tools.

giovannimanzoni, 2015-12-28:
Ok thanks for tell me what cross compiling tools are need for. But for example what is the complete step by step how to for make one image for BBB ? If I run crochet, che tell me that I need f reebsd suorce in /usr/src... What versione/revisione should I use ? Have I to do something in /usr/src before re run crochet script ?

giovannimanzoni, 2015-12-28:
update: I try again with your repo and FreeBSD 11 -r 291838 and all go ok, image created. I try to boot now but I do not understand why in the past I can not do that. I will reinstall pc-bsd for try to know why...

giovannimanzoni, 2015-12-28:
uhm.. files in boot partition are not right generated.. uboot stop with some error with 0x100 0x200 error on mmc... but if i substitute them with files present on http://raspbsd.org/bbb.html image, the sd card boot with my FreeBSD build. I will investigate why...

giovannimanzoni, 2016-01-07:
it work now. I clean my pcbsd installation and now it work.

closed 2016-01-07 by giovannimanzoni

jainformatyk, 2016-05-04:
Hello, I have identical error in my log when building kernel as in initial message for this thread. It tries ty find (..)/sys/MY_VERSATILECONF
when it is not there ofcourse but deeper in source tree 

crochet generated 
make TARGET_ARCH=armv6 SRCCONF=/dev/null __MAKE_CONF=/dev/null KERNCONF=MY_VERSATILECONF -DNO_CLEAN -DNO_ROOT -DKERNFAST -DNO_ROOT -j 1 buildkernel

Is it -j 1 to blame?

I also installed devel/gmake after I found this page. But after installation it still produces ths error message. I did not clean crochet work directory. Shall I?

jainformatyk, 2016-05-04:
@giovannimanzoni 
> it work now. I clean my pcbsd installation and now it work.
What do you mean? You cleaned /obj?

janjasiukowalski, 2016-06-04:
https://github.com/freebsd/crochet/issues/149