freebsd/crochet #139: Raspberry pi armv6hf

iyedb, 2016-01-01:
Not sure is the right place to ask or if it's even an issue but why raspberry pi board config does not set TARGET_ARCH to armv6hf? Why isn't it armv6hf by default.

kientzle, 2016-01-01:
Please send us a pull request with your suggested change.

iyedb, 2016-01-01:
#140 :)

zif2016, 2016-02-20:
can I add linux_base-c6 into RPi2 IMG?

kientzle, 2016-02-20:
You can try.

Step 1:  See if linux_base-c6 actually works on RPi2.  You should build and RPi2 image and try installing the package to see if it works (the FreeBSD port and package tools all work on RPi2).  If not, work with the FreeBSD developers to find out what would be needed to get it to work.

Step 2: Crochet has some experimental support for installing packages during the image creation.  They are documented in config.sample.sh.  They do not work for all packages because some packages cannot be cross-installed (the FreeBSD teams working on package support would like to support cross-installing more packages).

Step 3:  Let us know how it works for you.

zif2016, 2016-02-22:
I can't  get linux_base-c6 src in FreeBSD-arm v11 on RPi2. so that I never to make install it。

brd, 2016-06-08:
Closing.

For package installing see #141.

closed 2016-06-08 by brd