freebsd/crochet #141: option PackageInit fails

shepner, 2016-01-20:
So, when I add 'option PackageInit' to my config, I receive the following error:
```
Initializing package system
pkg: Warning: Major OS version upgrade detected.  Running "pkg-static install -f pkg" recommended
Updating FreeBSD repository catalogue...
pkg: http://pkg.FreeBSD.org/FreeBSD:11:armv6/latest/meta.txz: No address record
repository FreeBSD has no meta file, using default settings
pkg: http://pkg.FreeBSD.org/FreeBSD:11:armv6/latest/packagesite.txz: No address record
Unable to update repository FreeBSD
```
In futzing around, I found that manually running `sudo pkg-static -c ./work/_.mount.freebsd update` provides the same error but a wget (ie `sudo wget http://pkg.FreeBSD.org/FreeBSD:11:armv6/latest/meta.txz`) works fine.  And of course, manually running `pkg update` from the RPI itsel works just fine.  Im doing all of this from a bare bones FreeBSD 10.2 install

So Im going on the assumption that there is something wrong with pkg in regards to chroot but Im not sure how to work around it.  Alternatively, could I reasonably just compile ports directly or is that a path I dont want to travel?

thanks!
stephen

dicepd, 2016-05-22:
It took me a while to get past this same problem.
The main issue with this message is that pkg cannot resolve the address in a chroot environment.
To manually cure this add your /etc/resolv.conf to an overlay. These are inserted into the chroot before PackageInit.

A proper way to deal with this should be in PackageInit script which should do the following:
Check for /etc/resolv.conf.  (may have been placed by user in  overlay)
if exists  warn user that it must resolve for local network
if not exists then create two startegies:
The first to copy system /etc/resolve.conf to chroot env to be called before the PackageInit
The second to remove the chroot /etc/resolve.conf to be called after Package

This should save a few people some headaches trying to get around this problem

kientzle, 2016-05-22:
I'm not sure the suggestions made by dicepd are really appropriate.  For example, the user may have put an etc/resolv.conf into the overlay, but that will be appropriate for the target environment and may not be the same as used in the host environment.

I suspect that fixing this really requires changes to pkg itself to perform the necessary network lookups *before* it chroots.

bapt, 2016-05-22:
pkg supports an option for that which is pkg -o NAMESERVER=<ipofaserver> to avoid having to copy resolv.conf it was broken in 1.8 but I fixed it in pkg 1.8.0 which has been released today.

the problem with pkg -c is that it calls chroot very early. To avoid that problem we have added pkg -r <rootdir> which does not perform any chroot at all therefore having not network issue, but the ports tree are is not yet entirely aware of it and some scripts (preinstall/postinstall) might cause some issues.
at least creating users from the script is safe in that regard. for most simple case that should work.

brd, 2016-05-24:
The only problem with pkg -o NAMESERVER is you miss the search domains and stuff..  Is that a concern?

pr1ntf, 2016-08-01:
I've tried out adding a `resolv.conf` to my overlay and `pkg` works, it appears to want to install and update `pkg` from freebsd.org, not my specified repo, though.

pr1ntf, 2016-08-01:
Adding a `/usr/local/etc/pkg/repos/custom.conf ` overlay fixes the above problem.

kientzle, 2016-12-10:
@yohanesu75  has just sent some pull requests that might fix this.

Please take a look at PR #169, #170, and #171.

PR #189 (2017-03-19): Document pkg Features
Cleaned up the README a little bit and added some info about the `pkg` features in crochet. 

This effectively closes #141 I think?

Also, thanks @johalun for the sweet fix in #170!
closed 2017-03-19 by kientzle