freebsd/crochet #143: ZedBoard fail compilation

nanim93, 2016-02-18:
Hi everyone
Operating system: FreeBSD 9.3-RELEASE amd64

Source code in /usr/src: FreeBSD 11.0-CURRENT downloaded by the comand '_svn co "http://svn0.us-west.freebsd.org/base/head" /usr/src_'
svn co "http://svn0.us-west.freebsd.org/base/head" /usr/src
So, when I try to compile, I obtain this:
root@root: /bin/sh crochet.sh -c config_zedboard.sh
_Starting at Thu Feb 18 17:30:15 ART 2016
Loading configuration from config_zedboard.sh
Board: ZedBoard
Option: UsrPorts
        Will install up-to-date /usr/ports from portsnap
Option: UsrSrc
Source Version is: 295757
Building FreeBSD version: 11.0
Image name is:
        /home/FreeBSD_PC/Descargas/Crochet/work/FreeBSD-ZEDBOARD-ZedBoard-Src-Ports.img
Building FreeBSD version: 11.0
**eval: 1: Syntax error: redirection unexpected**_

I've used a config file that I made. But I obtained the same error if I use the generic config file by the comand: /bin/sh crochet.sh -b ZedBoard

Hernán Malatini

kientzle, 2016-02-19:
I suspect this is an issue with FreeBSD 9.  Do you have a newer FreeBSD system around?

kientzle, 2017-07-09:
Closing, since we don't officially support running Crochet on FreeBSD 9.

closed 2017-07-09 by kientzle