freebsd/crochet #144: Add Support for the Raspberry Pi 3

DatanoiseTV, 2016-02-29:
Differences to V2: http://raspberry.piaustralia.com.au/raspberry-pi-3-model-b-changes/

mko-io, 2016-06-08:
please the Freebsd arm64 image is in alpha and available for download:

ftp://ftp.freebsd.org/pub/FreeBSD/snapshots/arm64/aarch64/ISO-IMAGES/11.0//FreeBSD-11.0-ALPHA1-arm64-aarch64-20160528-r300895-memstick.img

rascal-hats, 2016-08-26:
11.0 links are dead, new links here:

ftp://ftp.freebsd.org/pub/FreeBSD/snapshots/arm64/aarch64/ISO-IMAGES/12.0/

http://ftp.freebsd.org/pub/FreeBSD/snapshots/arm64/aarch64/ISO-IMAGES/12.0/

catzilla4, 2016-10-29:
Sorry if I come off as a newbie, I'm interested in helping make the FreeBSD run on the Pi 3.  Unfortunately, I've been less than successful at it myself, so when I saw the initial commit I figured I should state this.

brd, 2016-10-30:
It has been committed and I have uploaded images to http://www.raspbsd.org.  There is still lots more work to be done.

closed 2016-10-30 by brd

mko-io, 2016-10-30:
@brd that's awesome. Do you have upload a image for Pine64?

brd, 2016-10-31:
@mko-io Uploaded to http://www.raspbsd.org/pine64.html !