freebsd/crochet #146: Crochet or snapshot image?

zkirill, 2016-03-26:
Is there a difference between images that are generated by crochet and images available on ftp://ftp.freebsd.org/pub/FreeBSD/snapshots/arm/armv6/ISO-IMAGES/11.0/?

brd, 2016-03-26:
Practically none..  Crochet just enables you to customize them at build time if you need.

zkirill, 2016-03-26:
@brd Thank you! I'm trying to install FreeBSD on a Banana Pi M1+ using a Kingston SDC10G2/32GB SD card and the image from FTP didn't boot on the BPI for some reason. The same BPI previously booted Bananian from the same SD card. I'm currently building using crochet to see if it books.

closed 2016-03-26 by zkirill

brd, 2016-03-26:
You might try imaging the SD card.  Sometimes they don't burn exactly right and won't boot.

zkirill, 2016-03-27:
@brd You were right! I think I messed up the burn command. Thanks again!