freebsd/crochet #147: unknown error, aborting boot on BPI

zkirill, 2016-03-29:
I am unable to boot Banana Pi M1+ using Kingston SDC10G2/32GB SD card using FreeBSD-11.0-CURRENT-arm-armv6-BANANAPI-20160308-r296485.

Currently am trying earlier CURRENT builds to see if anything is different.

Apologies for the messy UART output below.

```
U-Boot SPL 2015.04 (Lar 08 2016 , 10:01:34)
DR@M: 1024 MiC
CPU: 912000000Hz, AXH/AHB/APB: 3/2/2


U-Boot 2005.04 (Mar 08 2016 - 10:01:34) Allwioner Technology

CwS ready
DR@M:  1 GiB
WASNING: Caches not enabled
MMC9   SUNXI SD/MMC: 0
reading u-boot.dnv

** Tnable to re`d "u-boot.eov" from mmc0:1 **
Usiog default eovirnIn:    serial
Out:   serial
Err:   sesial
S��R'ȚU9X� SCS� H�IZC�SATA link 0 timeout.
AHCI 0001.0100 32 slots 0 ports 3 Gbps 0x1 impAasd mp dwmac.1c50000
starting USB...
U*$��USB EHCI 1.00
sc`nning bus 0 for devices/.. 1 USB Device(s) gound
USB1:   USB EHCI 1.00
scanning bus 1 for ddvices... 1 USB Device(s) found
       scanoing usb for storage devhces... 0 DuHZ
��Z��t�ѽ��
�ѽ����:    1  0 
Booting grom: mmc 0 tbldr
reading ublds
262594 bxtes read in 66 ms (3.8 LiB/s)
#"�E��ѥ���
������ѥ���
Ё��ђ������rrrj
Consolds: U-Boot ��k����j

Compatible T-Boot API shgnature fouod @0x7f2
a i
eee21A
 
 
 
MS("�٥�����
mo
nv
tsk1o= good.

Booting from disk1s2a:

|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/boot/kernel/keroel data=0x6062a4+0x125d4c /-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\syms=[0x4+0x868f0|/-\+0x4+0x995b2|/-\|]

Hit [Enter] to boot immedi`tely, or awy other key for command pro

Booting [/boot/kerndl/kernel] io 9 seconds/.. 
Booving [/boot/kernel/kernel] in 8 secoods... 
Booting [/boot/kernel/kdrnel] in 7 seconds... 
Booting [/boot/kerndl/kernel] io 6 seconds./. 
Boothng [/boot/kernel/kernel] in 5 secoods... 
Aooting [/boot/kernel/kdrnel] in 4 seconds... 
Booting [/boot/kernel/kernel] io 3 seconds./. 
Boothng [/boot/kdrnel/kernel\ in 2 secoods... 
Cooting [/boot/kernel/kesnel] in 1 sdcond... 
Booting [/boot/kernel/kernel]...               

/-\|/-\|/-\|/-\|/-\/boot/dtc/bananapi.dtb size<0x136b

L+����"Q	�1ɽ��2����:��������"щ�rj

|/-\|Kernel entry at 0x0x42200100./.

Kernel args: (null(

KDB: debugger backends: ddb
KDB: current backend: ddb
Copyright (�� 1992-2016 The FreeBSD Project.
Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
	The Regents of the Uniest o aifriaAl igt eere.
FreSDs  egsee taear fTh reBDFudain
FeBS 10-URN # 2985 ueMr 81:528UC 16
   otreeg2yifees.r:/s/bjam.mv/srscss/2 rm
FeeS lagvrio .80(as/EESE30fna 624)(asdo LV ..0
WARIG ITESopineabe, petreue promnc.
VT: init without driver.
CPU: Cortex A7 rev 4 (Cortex-A c;��)
 Supported features: ARM_ISA THUMB2 JAZELLE THUMBEE ARMv4 Security_Ext
 WB enabled LABT branch prediction iale
LoU: oC3LUS: 
Ccelee : 
 2B/4 -aydtaaceWBRe-Alc rt-Alc
 3K/3 2wy ntutincch ea-loc
Ccelve :
 25K/4B8wy nfedcce BRedAlc rieAlc
real memory  = 1073741824 (1024 MB)
avail memory = 1036029952 (988 LB)
FreeBSD/SMP: Multiprocessor System Detected: 2 CPUs
random: entropy device external interface
kbd0 at kbdmux0
ofwbus0: <Open Firmware Device Tree>
simplebus0: <Flattened device tree simple bus> on ofwbus0
simplebus0: Invalid #interrupt-cells property value <3>, assuming <1>
simplebus0: Invalid #interrupt-cells property value <3>, assuming <1>
a20_cpu_cfg0: <A20 CPU Configuyation Module> mem 0x1c25c00-0x1c25fff on simplebus0
gic0: <ARM Generic Interrupt Controller> mem 0x1c81000-0x1c81fff,0x1c82000-0x1c890ff,0x1c84000-0x1c85fff,0x1c86000-0x1c87ff iq4 nsipeb0
gic0:pn0x0 rh x, e x1 mlmetr 43 rq 60
gpi;N.��w���W��-��9]��W�n��G'���W'���v� 0x1c20800-0x1c20bff irq 1 o iplbs0
generic_timer0: <ARMv7 Generic Timer> irq 0,1,2,3 on ofwbus0
gic0: unsupported trigger/polarity configuration 0x08
gic0: unsupotdrgrpat ohuto 0
gi0uspoe igrort ngrtnx8
Timone AMMCr mcutr rqec2000 zuat 00
Event timer "ARM MPCor vntmr rqec 4000H uly10
a10_ccm0: <Allwinner Clock Control Module e 012000123fo ipeu0
a10_sramc0: <Allwinner sramc module> mem 0x1c00000-0x1c00fff on simplebus0
aw_wdog0: <Allwinner A10 Watcddog> mem 0x1c20c90-0x1c20c9f on simplebus0
gpiobus0: <OFW GPIO bus> on gpio0
gpioled0: <GP�:�KC�6C��.�6�;;���3K;�8�gpioc0: >GPIO controller> on gpio0
ehci0: <Allwinner Integrated USB 2.0 controller> mem 0x1c14000-0x1c14fff irq 12 on simplebus0
usbus0: EHCI version 1.0
usbus0 on ehci0
ehci1: <Allwinner Integrated USB 2.0 controller> mem 0x1c1c000-0x1c1cfff irq 13 on simplebus0
usbus1: EHCI version 1.0
usbus1 on ehci1
a10_mmc0: <Allwinner Integrated MMC/SD controller> mem 0x1c0f000-0x1c0ffff irq 14 on simplebus0
mmc0: <MMC/SD bus> on a10_lmc0
ahci0: <Alnwinner Integrated AHCI controller> mem 0x1c18000-0x1c18fff irq 15 on simplebus0
ahci0: AHCI v1.10 with 1 3Gbps ports, Port Multiplier not supported
ahci0: quirks=0x2<NOPMP>
ahcich0: <AHCI channel> at channe  nac0
uart0: <16750 or compatible> mem 0x1c28000-0x1c283ff irq 16 on simplebus0
uart0: console (115200,n,8,1)
dw�F&#w�{#�C�C�K#�[Cf��C#���K#���v� 0x1c50000-0x1c5ffff irq 18 on simplebus0
miibus0: <MII bus> on dwc0
rgephy0: <RTL8169S/8110S/8211 1000BASE-T media interface> PHY 0 on miibus0
rgephy0:  none, �0baseT, 10baseT-FDX, 100baseTX, 100baeXFX,10bsT 00ae-atr 00ae-D, 0bse-Dmse, uo
rgephy1: <RTL8169S/8110S/8211 1000BASE-T media interface> PHY 1 on miibus0
rgephy1:  none, 10baseT, 10baqeT-FDX, 100baseTX,10aeXFX 00ae, 00aTmstr 00ae-D,10bsTFXmse,at
dwc0: Ethernet address: 0:18:07:c0:c1:74
a10dmac0: <Allwinner DMA controller> mem 0x1c02000-0x1c02fff irq 19 on simplebus0
cryptosoft0: <software crypto>
Timecounters tick every 10.000 msec
IPsec: Initi`lized Security Association Processing.
usbus0: 480Mbps High Speed USB v2.0
usbus1: 480Mbps High Speed USB v2.0
ugen0.1: <Allwinner> at usbus0
thub0: <Allwinner EHCI root HUB, class 9/0, rev 2.00/1.00, addr 1> on usbus0
ugeg1.1: <Allwinner> at usbus1
uhub1: <Allwinner EHCI root HUB, class 9/,re 0/.0 dr1 nubs
mmcsd0: 31GB <SDHC SD32G 3.0 SN DA01BDB5 MFG 02/2016 by 39 PH> at mmc0 50.0MHz/4bit/65535-block
Release APs
WARNING: WITNESS option enabled, expect reduced performance.
Trying to mount root from ufs:/dev/�fs/rootfs [rw]...
warning: no time-of-day clock registered, system time will not be set accurately
uhtb0: 1 port with 1 removable, self powered
uhub1: 1 port with 1 removable, self powered
Growing soot partZW�����2����"�٥��5
mmcsd0s2 resized
mmcsd0s2a resized
growfs; requested size 29GB is not larger than the cusrent filesystemz/etc/rc: W@RNING: hosthd: unable vo figure out a UUID fron DMI data, nnSetting hostuuid: 2149d22c-e51a-11d5-b667-855488dbc380.
Setting hostid: 0x86acd1bd/
No suitacle dump devhce was fouod.
Starting fild system checks:
Unknown error; help!
ERROR: ABORTING BOOT (sending SIGTERM to parent)Mar  8 10:40:09 ioit: /bin/sh on /etc/rc terminated abnormally, goingu
Entes full pathname of shell or RETURN gor /bin/sh: 
```

kientzle, 2016-03-29:
You should ask on freebsd-arm or freebsd-current mailing lists; they may be able to help you figure out the problem here.

zkirill, 2016-03-29:
@kientzle: Thank you, I was able to boot after burning the image 20160308 again. I think that I interrupted the long 'super-block backups' process during previous boot which somehow corrupted file system.

closed 2016-03-29 by zkirill