freebsd/crochet #149: freebsd/sys/PANDABOARD: No such file or directory *** [buildkernel] Error code 2

janjasiukowalski, 2016-06-04:
My pandaboad.sh file

```
#
# SAMPLE CONFIGURATION
#
# Copy this to another file and use that when running Crochet, e.g.
#  $ sudo /bin/sh crochet.sh -c <myconfig>
#

# Note: Only board_setup is actually required, assuming /usr/src has
# suitable FreeBSD sources.  Everything else below is optional.

# REQUIRED: 
# Uncomment one to choose the default configuration for your board.
#
# THIS MUST BE THE FIRST LINE IN YOUR CONFIG.SH FILE!
#
# Don't see your board here?  board/NewBoardExample has details for
# how to add a new board definition.  If you need help, ask.  If you
# get it working, please consider contributing it.
#
# Read board/<board-name>/README for more details
# about configuring for your particular board.
#
#board_setup Alix
#board_setup BananaPi
#board_setup BeagleBone
#board_setup Cubieboard2
#board_setup Chromebook
#board_setup GenericI386
board_setup PandaBoard
#board_setup RadxaRock
#board_setup RadxaRockLite
#board_setup Soekris
#board_setup RaspberryPi
#board_setup RaspberryPi2
#board_setup VersatilePB
#board_setup ZedBoard
#board_setup Wandboard
#

# Size of the disk image that will be built.  This is usually the same
# size as your memory card or disk drive, but it can be smaller.
# Each board setup above defines a default value, but the default is
# deliberately chosen to be just barely big enough to hold a minimal
# FreeBSD system.
#
# 'mb' = 10 ^ 6,  'gb' = 10 ^ 9, 'mib' = 2 ^ 20 and 'gib' = 2 ^ 30
#
# From 2%-10% of the flash on a typical memory card is
# reserved for the hardware controller, so the filesystem
# image size needs to be a bit smaller than the card size:
#
# Suggested: option ImageSize
#option ImageSize 100mb # for kernel-only images
#option ImageSize 1950mb # for 2 Gigabyte card
#option ImageSize 3900mb # for 4 Gigabyte card

#
# How to Customize Your Build
#
# Crochet supports a variety of customization approaches:
#
# 1) "options" provide canned functionality.  If there is not
#   an option that provides what you want, consider contributing
#   a new one.  There's a template in option/Sample that explains
#   how to write a new one.  Note that board definitions can include
#   their own options; a few of the options below are only defined
#   for certain boards.
[... 316 lines cut ...]
--------------------------------------------------------------
>>> stage 2.3: build tools
--------------------------------------------------------------
cd /usr/home/localuser/PROJECT/freebsd; TOOLS_PREFIX=/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp MAKEOBJDIRPREFIX=/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp  INSTALL="sh /usr/home/localuser/PROJECT/freebsd/tools/install.sh"  PATH=/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/legacy/usr/sbin:/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/legacy/usr/bin:/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/legacy/bin:/sbin:/bin:/usr/sbin:/usr/bin  WORLDTMP=/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp /usr/home/localuser/PROJECT/crochet/work/obj/usr/home/localuser/PROJECT/freebsd/make.i386/bmake  -f Makefile.inc1  DESTDIR=  BOOTSTRAPPING=1001000  SSP_CFLAGS=  MK_HTML=no -DNO_LINT MK_MAN=no  -DNO_PIC MK_PROFILE=no -DNO_SHARED  -DNO_CPU_CFLAGS MK_WARNS=no MK_CTF=no kernel-tools
--- kernel-tools ---
mkdir -p /usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/usr
mtree -deU -f /usr/home/localuser/PROJECT/freebsd/etc/mtree/BSD.usr.dist  -p /usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/usr >/dev/null
--------------------------------------------------------------
>>> stage 3.1: building everything
--------------------------------------------------------------
cd /usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/sys/PANDABOARD; MAKEOBJDIRPREFIX=/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6  MACHINE_ARCH=armv6  MACHINE=arm  CPUTYPE= GROFF_BIN_PATH=/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/legacy/usr/bin  GROFF_FONT_PATH=/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/legacy/usr/share/groff_font  GROFF_TMAC_PATH=/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/legacy/usr/share/tmac CC="cc " CXX="c++  "  CPP="cpp "  AS="as" AR="ar" LD="ld" NM=nm  OBJDUMP=objdump OBJCOPY="objcopy"  RANLIB=ranlib STRINGS=  SIZE="size"  INSTALL="sh /usr/home/localuser/PROJECT/freebsd/tools/install.sh"  PATH=/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/legacy/usr/sbin:/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/legacy/usr/bin:/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/legacy/bin:/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/usr/sbin:/usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/tmp/usr/bin:/sbin:/bin:/usr/sbin:/usr/bin /usr/home/localuser/PROJECT/crochet/work/obj/usr/home/localuser/PROJECT/freebsd/make.i386/bmake  -D NO_CLEAN -D KERNFAST -D NO_ROOT -j 2 -J 15,16 -m /usr/home/localuser/PROJECT/freebsd/share/mk  KERNEL=kernel all -DNO_MODULES_OBJ
cd: /usr/home/localuser/PROJECT/crochet/work/obj/arm.armv6/usr/home/localuser/PROJECT/freebsd/sys/PANDABOARD: No such file or directory
*** [buildkernel] Error code 2

bmake[1]: stopped in /usr/home/localuser/PROJECT/freebsd
1 error

bmake[1]: stopped in /usr/home/localuser/PROJECT/freebsd
*** [buildkernel] Error code 2

make: stopped in /usr/home/localuser/PROJECT/freebsd
1 error

make: stopped in /usr/home/localuser/PROJECT/freebsd
```

kientzle, 2016-06-05:
Please try without using these options:

FREEBSD_INSTALLWORLD_EXTRA_ARGS="-DNO_ROOT "
FREEBSD_BUILDKERNEL_EXTRA_ARGS="-DNO_ROOT "
FREEBSD_INSTALLKERNEL_EXTRA_ARGS="-DNO_ROOT "
FREEBSD_EXTRA_ARGS="-DNO_CLEAN"
FREEBSD_KERNEL_EXTRA_ARGS="-DKERNFAST"

janjasiukowalski, 2016-06-12:
OK, now it works.

closed 2016-06-12 by janjasiukowalski