freebsd/crochet #156: Build stops at MK_MAN :

sapiens-sapide, 2016-07-07:
Hi, 
I tried to build an image for the wandboard-quad. The script (launched with root user) ran 30 minutes and stopped with the following message : 

```
(…)
Building FreeBSD armv6-IMX6 ubldr at jeu 7 jul 2016 06:12:18 CEST
     (Logging to /tmp/crochet/work/ubldr-armv6-IMX6/_.ubldr.armv6-IMX6.build.log)
make: "/usr/src/share/mk/bsd.own.mk" line 480: MK_MAN can't be set by a user.
```

Any clue ?
Thanks,
Stan.

brd, 2016-07-07:
What svn rev are you trying to build and from what branch?

sapiens-sapide, 2016-07-08:
I git clone the repository yesterday. So it's branch `master` at commit `69887b0ec1`.
I tried to build `/board/Wandboard-Quad` with option `option User` activated.

brd, 2016-08-17:
Hmm, this should be fixed in 69887b0ec1080ad8f25c19aa4fcc7cd414f58949.  Does git log show you have that commit?

brd, 2016-10-15:
I think your /usr/src is out of date, but since there has been no reply I am going to close this out.

closed 2016-10-15 by brd