freebsd/crochet #158: Couple of questions

brd, 2016-08-17:
@kientzle These might be best directed at you / I would appreciate your input.

1. In https://github.com/freebsd/crochet/blob/master/lib/scm.sh#L34 svnversion is called to figure out the SVN revision.  But in another place we just pull it from the output of `svn info`, which is much cheaper.  Would it be helpful to switch to that?

2. We also do not rebuild world if the svn rev or git hash has changed.  I think it would be useful to do so.  In my scripts to build images I have added a `rm -fr $OJBDIR` to force crochet to do a build.

kientzle, 2016-08-17:
> In https://github.com/freebsd/crochet/blob/master/lib/scm.sh#L34 svnversion is called to figure out the SVN revision. But in another place we just pull it from the output of svn info, which is much cheaper. Would it be helpful to switch to that?

`svn info` is cheaper because it’s less correct.  `svn info` only reports the version of the root directory; `svnversion` returns a range of versions that accounts for the possibility that some parts of the tree might be newer than others.  (Remember that svn explicitly supports mixed/partial checkouts.)  To see the difference, try `svn update sys && svnversion`.

kientzle, 2016-08-17:
> We also do not rebuild world if the svn rev or git hash has changed. I think it would be useful to do so. In my scripts to build images I have added a rm -fr $OJBDIR to force crochet to do a build.

Yes, it should.  I suggest writing the svn rev or git hash into $OBJDIR somehow to mark that it was built from a particular revision.  Then you could check that revision marker using code similar to:

```
echo “$SOURCE_VERSION” > $WORKDIR/_objdir_revision
if [ -e “$OBJDIR/_crochet_revision” ] && diff $WORKDIR/_objdir_revision $OBJDIR/_crochet_revision; then
   # Diff succeeded, same revision
else
   # Diff failed or revisions different
   cp $WORKDIR/_objdir_revision $OBJDIR/_crochet_revision
   # … rebuild ...
fi 
```

brd, 2016-08-17:
1. Would you mind if I switched it to use `svn info`?  I don't think most users are only updating part of their tree.

2. OK, I will work on stashing the rev/hash somewhere so we can take advantage of it.

3. I will work on this, since 9.x EOL is in a few months.

4. With all of @bdrewery's work in HEAD & 11 around incremental builds, I think if we have a /dev/filemon we should do incremental builds by default.  Thoughts?

bdrewery, 2016-08-17:
Regarding 4 that means setting 'WITH_META_MODE=yes' in the environment.

brd, 2016-08-17:
@bdrewery Yep.