freebsd/crochet #159: RPI2: Failed to build FreeBSD armv6 world & IPFW issue

prov3it, 2016-09-13:
Im stuck. I've been using crochet for a week now. I've build some images but always found something to add to the src- and make.conf. This time im getting an error when i want to build world. I cant find any related errors on the web. How can i solve this?

Also i've tried adding ipfw-support by adding these lines to: src/sys/arm/conf/RPI2.
```
options    IPFIREWALL			# enables IPFW
options    IPFIREWALL_VERBOSE		# enables logging for rules with log keyword
options    IPFIREWALL_VERBOSE_LIMIT=5	# limits number of logged packets per-entry
options    IPFIREWALL_DEFAULT_TO_ACCEPT # sets default policy to pass what is not explicitly denied
**options    IPDIVERT			# enables NAT**
``` 

However, i noticed that i got errors while compiling the kernel because of the last option. should it still be possible in FreeBSD 12 to add ipfw support this way? do note that the error i wrote about initially in the first paragraph did not include these options.

make.conf:
```
BOOTWAIT=0
MALLOW_PRODUCTION=1
NO_SHARE=1
BUILD_OPTIMIZED=YES
NO_SENDMAIL=TRUE
NO_GAMES=TRUE
NO_BIND=TRUE
NO_SHARE=TRUE
NO_TCSH=TRUE
NO_OPENSSL=TRUE
NO_CRYPT=TRUE
NO_BLUETOOTH=TRUE
NO_MAILWRAPPER=TRUE
NO_INET6=TRUE
NO_ATM=TRUE
NO_FORTRAN=TRUE
NO_GDB=TRUE
NO_SENDMAIL=TRUE
```

src.conf:
```
WITHOUT_ATM=YES
WITHOUT_AMD=YES
WITHOUT_APM=YES
WITHOUT_CTM=YES
WITHOUT_DICT=YES
WITHOUT_FINGER=YES
WITHOUT_FREEBSD_UPDATE=YES
WITHOUT_GPIO=YES
WITHOUT_INSTALLLIB=YES
WITHOUT_KERBEROS=YES
WITHOUT_KVM_SUPPORT=YES
WITHOUT_NETGRAPH=YES
WITHOUT_PAM_SUPPORT=YES
WITHOUT_PROFILE=YES
WITHOUT_TFTP=YES
WITHOUT_VT=YES
WITHOUT_BLUETOOTH=YES
WITHOUT_FLOPPY=YES
WITHOUT_GAMES=YES
WITHOUT_HYPERV=YES
WITHOUT_INET6=yes
WITHOUT_INET6_SUPPORT=yes
WITHOUT_MAIL=YES
WITHOUT_MAN=YES
WITHOUT_PAM=YES
WITHOUT_PAM_SUPPORT=YES
WITHOUT_PPP=YES
WITHOUT_RADIUS_SUPPORT=YES
WITHOUT_SENDMAIL=YES
WITHOUT_TCSH=YES
WITHOUT_TELNET=YES
WITHOUT_UTMPX=YES
WITHOUT_WIRELESS=YES
[... 148 lines cut ...]

bmake[2]: stopped in /usr/home/crochet-master/src
--- _bootstrap-tools-lib/clang/libllvmsupport ---
A failure has been detected in another branch of the parallel make

bmake[3]: stopped in /usr/home/crochet-master/src/lib/clang/libllvmsupport
*** [_bootstrap-tools-lib/clang/libllvmsupport] Error code 2

bmake[2]: stopped in /usr/home/crochet-master/src
4 errors

bmake[2]: stopped in /usr/home/crochet-master/src
*** [_bootstrap-tools] Error code 2

bmake[1]: stopped in /usr/home/crochet-master/src
1 error

bmake[1]: stopped in /usr/home/crochet-master/src
*** [buildworld] Error code 2

make: stopped in /usr/home/crochet-master/src
1 error

make: stopped in /usr/home/crochet-master/src
```

xmj, 2017-04-06:
Is this on FreeBSD -CURRENT?

kientzle, 2017-07-09:
The freebsd-arm or freebsd-current mailing lists would probably be a better place to ask this:

  https://www.freebsd.org/community/mailinglists.html

closed 2017-07-09 by kientzle