freebsd/crochet #160: Generic I386 board file build

frankharvey, 2016-09-15:
I have failure of Generic I386 board file build. It fails in the end at bsdlabel.
Is this something I have done wrong? I used a custom setup.sh and used the crochet.sh -b <GenericI386> method.
Is the I386 board file working for others? Here is a snip of my error.

`    Object files are at: /usr/local/src/crochet/work/obj/i386.i386/usr/src
    Found suitable FreeBSD source tree in:
      /usr/src
    Using FreeBSD i386 world from previous build
    Using FreeBSD i386-GENERIC kernel from previous build
    Creating a 600MB raw disk image in:
      /usr/local/src/crochet/work/FreeBSD-i386-11-GENERIC.img
    Partitioning the raw disk image with MBR at Wed Sep 14 05:24:25 EDT 2016
    gpart create -s MBR md1
    md1 created
    Creating an auto-sized UFS partition at Wed Sep 14 05:24:25 EDT 2016
    md1s1 created
    /dev/md1s1a: 572.1MB (1171712 sectors) block size 32768, fragment size 4096
      using 4 cylinder groups of 143.06MB, 4578 blks, 18432 inodes.
    super-block backups (for fsck_ffs -b #) at:
     192, 293184, 586176, 879168
    tunefs: soft updates set
    Using inode 4 in cg 0 for 4194304 byte journal
    tunefs: soft updates journaling set
    tunefs: NFSv4 ACLs set
    Installing bootblocks
    Boot files are at: /usr/local/src/crochet/work/obj/i386.i386/usr/src/sys/boot/i386
    bootcode written to md1
    active set on md1s1
    bsdlabel: /dev/md1s1a: no valid label found
    root@DEV:/usr/local/src/crochet #
`
I have built RPi2 and BBB images successfully.

frankharvey, 2017-05-08:
I copied a few sections from the Alix file to get this working. Can we please add this.
**/crochet/board/GenericI386/setup.sh**

https://gist.github.com/frankharvey/a26b5a682b4efb09878e449606d69304

kientzle, 2017-07-09:
Please take a look at 008fe52 and let me know if it works for you.

closed 2017-07-09 by kientzle