freebsd/crochet #161: Support for Orange Pi One

derneue, 2016-10-11:
What Needs to be done for supporting the Orange Pi one? It is a low cost Board (15$ on Aliexpress) with an Allwinner H3 SoC which is supported by FreeBSD. An uboot-port for this board is also available  in sysutils/u-boot-orangepi-one.

If someone has a Setup.sh file to test, I will receive Hardware in 2 weeks.

evadot, 2016-10-15:
Duplicate of issue #157

CurtisVillamizar, 2018-05-08:
For now see http://occnc.com/info/arm64/NOTES.html .   Will be creating a few pull requests but currently have and issue with github itself.  The notes file is ahead of reality in that I haven't yet had time to test some of the ports I've built on Orange Pi One, just Orange Pi PC Plus.

CurtisVillamizar, 2018-05-11:
I replaced that with a cleaner pull request (the other had fixes for other boards that snuck in).
https://github.com/freebsd/crochet/pull/246

CurtisVillamizar, 2018-06-06:
Someone please close this.  A board directory now exists since pull request https://github.com/freebsd/crochet/pull/242 was picked up.

laytonwang, 2018-06-08:
I think this topic should be kept.

Succeed in compiling on FreeBSD 12.0-CURRENT #0 r334337(released on Tue May 29 17:12:35 UTC 2018) and set target to armv6 or armv7 by "-b" parameter .
But hung when booting above images either armv6 or armv7 , error message is the following:
Loaded DTB from file 'sun8i-h3-orangepi-one.dtb'
kernel entry at 0x42200100...
kernel args: (null)

Any successful runing? I am appreciated if anyone may let me know the cause of this error or your detailed and correct system environment . BTW , I couldn't get FreeBSD 12.0-CURRENT r333094 as mentioned in "http://occnc.com/info/arm64/NOTES.html" above.

Thanks!