freebsd/crochet #163: Missing files for Allwinner kernel build

felix, 2016-10-26:
Building for Banana Pi currently fails while building kernel with the following:
```
>>> stage 3.1: building everything
--------------------------------------------------------------
...
cd: /usr/home/felix/src/crochet/work/obj/arm.armv6/usr/home/felix/src/crochet/src/sys/ALLWINNER: No such file or directory
*** [buildkernel] Error code 2
```
World seems to build fine. My FreeBSD source is on stable/11 branch at 2b54e88 and crochet is at current HEAD of 7feec7a.

kientzle, 2016-10-27:
The kernel build process first "configures" a kernel by building a directory full of compile information, then changes into that directory where the actual kernel build is performed.

I believe the error you list is saying that it can't find the directory that should have been created by the configure process.  What are the contents of

 `/usr/home/felix/src/crochet/work/obj/arm.armv6/usr/home/felix/src/crochet/src/sys`

Were there any earlier errors reported when it tried to configure the kernel?

felix, 2016-10-27:
No previous errors.
```
$ pwd
/home/felix/src/crochet/work/obj/arm.armv6/usr/home/felix/src/crochet/src/sys
$ tree -d
.
└── boot
    ├── arm
    │   └── uboot
    ├── common
    ├── efi
    │   ├── boot1
    │   │   └── machine -> /usr/home/felix/src/crochet/src/sys/boot/efi/boot1/../../../arm/include
    │   ├── fdt
    │   │   └── machine -> /usr/home/felix/src/crochet/src/sys/boot/efi/fdt/../../../arm/include
    │   ├── libefi
    │   └── loader
    │       └── machine -> /usr/home/felix/src/crochet/src/sys/boot/efi/loader/../../../arm/include
    ├── fdt
    ├── ficl
    ├── forth
    └── uboot
        ├── fdt
        │   └── machine -> /usr/home/felix/src/crochet/src/sys/boot/uboot/fdt/../../../arm/include
        └── lib
            └── machine -> /usr/home/felix/src/crochet/src/sys/boot/uboot/lib/../../../arm/include
```

frankharvey, 2016-11-05:
I just ran into this problem. It appears to me that the source bananapi.dts is broke. 
When I run DTC I get this:
dtc -I dts -o dtb bananapi.dts > /bananapi.dtb
Error:bananapi.dts:29.1-2 syntax error
FATAL ERROR: Unable to parse input tree

I get no such failures with BBB and RPi2 and the DTC

frankharvey, 2016-11-05:
The bananapi.dts references on line 29
#include "sun7i-a20-bananapi.dts

This file does not exsist in the FreeBSD FDT/DTS source folder but does reside in the GNU/DTS source folder.
Simply copying it over does not help. It needs fixup for FreeBSD I guess.

evadot, 2016-11-05:
dts files have to be pre-processed by cpp, see sys/tools/fdt/make_dtb.sh

frankharvey, 2016-11-05:
Thanks
That is strange as I used dtc for RPi2 and it generated an DTB. I was using it to check my syntax before crochet usage for adding DS3231.

I am sure you are correct though as I deleted the line and it errored the next #include line.

That is good as I thought my RTC device tree code should be correct. I will try and pre process it.

frankharvey, 2016-11-06:
I ran a fresh crochet built via ./crochet.sh -BananaPi
Here is the tail of the error log on FreeBSD11p1
`--- all_subdir_de ---
--- machine ---
machine -> /usr/src/sys/arm/include
--- opt_ddb.h ---
ln -sf /crochet/work/obj/arm.armv6/usr/src/sys/ALLWINNER/opt_ddb.h opt_ddb.h
--- all_subdir_dtb/allwinner ---
===> dtb/allwinner (all)
--- all_subdir_de ---
--- device_if.h ---
awk -f /usr/src/sys/tools/makeobjops.awk /usr/src/sys/kern/device_if.m -h
--- all_subdir_dtb/allwinner ---
--- bananapi.dtb ---
Generating bananapi.dtb from bananapi.dts
converting bananapi.dts -> /crochet/work/obj/arm.armv6/usr/src/sys/ALLWINNER/modules/usr/src/sys/modules/dtb/allwinner/bananapi.dtb
--- all_subdir_de ---
--- bus_if.h ---
awk -f /usr/src/sys/tools/makeobjops.awk /usr/src/sys/kern/bus_if.m -h
--- all_subdir_dtb/allwinner ---
Error: <stdin>:1.1-2 syntax error
FATAL ERROR: Unable to parse input tree
*** [bananapi.dtb] Error code 1

make[4]: stopped in /usr/src/sys/modules/dtb/allwinner
1 error

make[4]: stopped in /usr/src/sys/modules/dtb/allwinner
*** [all_subdir_dtb/allwinner] Error code 2

make[3]: stopped in /usr/src/sys/modules
--- all_subdir_de ---
A failure has been detected in another branch of the parallel make
`

frankharvey, 2016-11-06:
You are correct and I did build crochet.sh -b BananaPi without any trouble.

felix, 2016-11-22:
This is not an issue with stable/11, which I failed to mention.

closed 2016-11-22 by felix