freebsd/crochet #164: Building XDev To Build U-Boot

ThothK, 2016-11-05:
Hello People,

I have been having problems with building the Cross Development tools.

Some of the variants I tried include:
- make XDEV=arm XDEV_ARCH=armv6 xdev

- make XDEV=arm XDEV_ARCH=armv6 xdev-build

- make XDEV=arm XDEV_ARCH=armv6 WITH_GCC=1 WITH_GCC_BOOTSTRAP=1 WITHOUT_CLANG=1 WITHOUT_CLANG_BOOTSTRAP=1 WITHOUT_CLANG_IS_CC=1 xdev

- make XDEV=arm XDEV_ARCH=armv6 WITH_GCC=1 WITH_GCC_BOOTSTRAP=1 WITHOUT_CLANG=1 WITHOUT_CLANG_BOOTSTRAP=1 WITHOUT_CLANG_IS_CC=1 xdev xdev-links

- make XDEV=arm XDEV_ARCH=armv6 WITH_GCC=1 WITH_GCC_BOOTSTRAP=1 WITHOUT_CLANG=1 WITHOUT_CLANG_BOOTSTRAP=1 WITHOUT_CLANG_IS_CC=1 WITHOUT_TESTS=1 xdev

With one exception stddef.h is not found.
===> lib/libc++ (obj,all,install)
/usr/obj/armv6-freebsd/usr/src/lib/libc++ created for /usr/src/lib/libc++
...
In file included from /usr/src/lib/libc++/../../contrib/libc++/src/algorithm.cpp:10:
In file included from /usr/src/lib/libc++/../../contrib/libc++/include/algorithm:624:
In file included from /usr/src/lib/libc++/../../contrib/libc++/include/initializer_list:47:
/usr/src/lib/libc++/../../contrib/libc++/include/cstddef:43:15: fatal error:
      'stddef.h' file not found
#include_next <stddef.h>

The one exception I mentioned was a linker error. However, the above is what I received most of the time.

I have tried with FreeBSD versions 10.0, 10.3, and 11.0.1. All fresh installs as Virtual Machines.

If I understand correctly, U-Boot has to be compiled with gcc and we need an ARM 'version' of gcc to compile U-Boot for our purposes. 
However, I seem to be unable to build the Cross Development tools.

This seems to be a FreeBSD problem and I should let someone know, but I do not know who nor how.

I would appreciate help and / or suggestions. I really would like to be able to compile U-Boot in the FreeBSD VM so I can try pushing it to a device.

=thoth=

brd, 2016-11-05:
Hi,

IIRC xdev is deprecated.  Why are you using it?  What ARM board are you building it for?  We have ports for a lot of them that you should use instead if possible.

kientzle, 2016-11-05:
> If I understand correctly, U-Boot has to be compiled with gcc and we need an ARM 'version' of gcc to compile U-Boot for our purposes.

Most Crochet boards now use U-Boot from ports, so Crochet should no longer need to compile U-Boot at all.  In particular, xdev should not be needed.

ThothK, 2016-11-06:
Hi Brad,

I did not realize that XDev was deprecated. I thought it was no longer required. Even still, I should have been able to build the tools with FreeBSD 10.3, right?

I was trying to get the Cross Compile tools to build U-Boot. This is the only way I know how on FreeBSD.

I am building it for an NVidia TK1 (ARM Cortex-A15) in an NVidia Shield K1 tablet.

I did not see Jetson or any of the other NVidia products in the boards Crochet boards list. Did I miss it? I will slap the back of my head if you did, but still. 
Is there someplace other than the supported boards list I should be looking?

I have the U-Boot source that NVidia is using for their Ubuntu distribution. Since I know that works, I thought I I should determine how to build it, find out the changes Crochet makes (used to make), implement those, and build it again with the Crochet changes or whatever is appropriate.

=thoth=

ThothK, 2016-11-06:
Hi Tim,

Do you mean U-Boot from FreeBSD Ports? Should this be in my ports tree in the FreeBSD VM? 
If so, I did not make that connection after reading /freebsd/crochet/blob/master/lib/uboot.sh or anything else. 

You have answered a different question I have been trying to find the answer for. That is that Crochet does not need to compile U-Boot for the most part.

I have the U-Boot source that NVidia is using for their Ubuntu distribution. Since I know that works, I thought I I should determine how to build it, find out the changes Crochet makes (used to make), implement those, and build it again with the Crochet changes or whatever is appropriate.

=thoth=

ThothK, 2016-11-06:
Hi,

Since there is a different path now, does anyone have any suggestions on how I should compile U-Boot on FreeBSD? Unless I can find a FreeBSD ready version of U-Boot NVidia is using for their Ubuntu distro, I think I need to learn how to build it on FreeBSD.

If there is something obvious that I am missing or a different tack I should take, please let me know.
The objective is to get FreeBSD running on the NVidia Shield K1 tablet.

=thoth=

kientzle, 2016-11-06:
> Do you mean U-Boot from FreeBSD Ports?

Yes.

> Should this be in my ports tree in the FreeBSD VM? 

Yes.

> If so, I did not make that connection after reading /freebsd/crochet/blob/master/lib/uboot.sh or anything else.

That code is legacy code and should probably be removed.  At the time it was written, the ports tree did not have working cross-compilers for ARM, so xdev was the best available option.  Since then, the ports tree now has solid cross-compilers available, xdev is no longer supported, and there are U-Boot ports for the most popular boards.

> suggestions on how I should compile U-Boot on FreeBSD?

Start by copying one of the existing U-Boot ports.  They use cross-compiler tools from ports, which are stable and supported now.  Best of all, if you can get a new port working, you can submit it to the FreeBSD ports collection for others to use.

ThothK, 2016-11-06:
Hi Tim,

Thank you. 
I am going to close this issue since building the cross compiler tools is no longer part of the process.

I intend on submitting whatever I can get working that someone has not already done. If this effort pays off and I can get the company I work for to switch to FreeBSD for the project, and we make improvements or whatever, I intend on pushing to submit that as well.

This just reminds me about how much I have to learn about FreeBSD.

=thoth=

closed 2016-11-06 by ThothK

brd, 2016-11-06:
Great!  We would welcome your contributions to getting FreeBSD to work on another ARM device.  It can be difficult to get drivers and things written, but I wish you luck!

brd, 2016-11-06:
Oh, I forgot to mention you should email the freebsd-arm list and ask if anyone else has that hardware.