freebsd/crochet #167: alix on 12 is not building/booting

fbettag, 2016-11-21:
Hi,

i've been trying to get an alix image running HEAD to working for two days now. My PR from yesterday fixes the build process to a certain point, but whatever i specify as disk-size in config.sh, i get these:

Creating a 1950MB raw disk image in:
    /root/crochet-master/work/FreeBSD-i386-12.0-ALIX_el8HOME-308927.img
Partitioning the raw disk image with MBR at Mon Nov 21 19:52:02 CET 2016
gpart create -s MBR md0
md0 created
Creating an auto-sized UFS partition at Mon Nov 21 19:52:02 CET 2016
md0s1 created
/dev/md0s1a: 1858.9MB (3807104 sectors) block size 32768, fragment size 4096
       	using 4 cylinder groups of 464.75MB, 14872 blks, 59520 inodes.
super-block backups (for fsck_ffs -b #) at:
 192, 952000, 1903808, 2855616
tunefs: soft updates set
Using inode 4 in cg 0 for 4194304 byte journal
tunefs: soft updates journaling set
tunefs: NFSv4 ACLs set
Installing bootblocks
Boot files are at: /root/crochet-master/work/obj/i386.i386/usr/src/sys/boot/i386
bootcode written to md0
active set on md0s1
partition c: partition extends past end of unit
bsdlabel: partition c doesn't cover the whole unit!
bsdlabel: An incorrect partition c may cause problems for standard system utilities
=>     63  3808530  md0  MBR  (1.8G)
       63      961       - free -  (481K)
     1024  3807232    1  freebsd  [active]  (1.8G)
  3808256      337       - free -  (169K)

The resulting image is unbootable, "No Operating System", when dd'd to a CFdisk. When i run bsdlabel and gpart bootcode manually after crochet is done, it at least starts BTX Loader and boots the kernel, but somehow stops in the middle of the boot (i think right before it's supposed to mount the disk).

Any ideas or pointers welcome.

thanks

roycewilliams, 2018-02-27:
I have a similar problem, but the failure happens at a slightly earlier point (gpart MBR stage):
```
Starting at Mon Feb 26 07:20:20 AKST 2018
Loading configuration from config.sh
Board: Alix
Option: ImageSize 3900mb
Option: Growfs
Option: Email royce@tycho.org
Option: User royce
Source version is: unknown
Building FreeBSD version: 11.1
Copying build config ALIX11 to source tree
Image name is:
    /usr/local/src/hw/crochet/work/FreeBSD-i386-11-ALIX-Alix.img
Building FreeBSD version: 11.1
Object files are at: /usr/local/src/hw/crochet/work/obj/i386.i386/usr/src
Found suitable FreeBSD source tree in:
    /usr/src
Building FreeBSD i386 world at Mon Feb 26 07:20:21 AKST 2018
    (Logging to /usr/local/src/hw/crochet/work/_.buildworld.i386.log)
Building FreeBSD i386-ALIX kernel at Mon Feb 26 15:16:54 AKST 2018
    (Logging to /usr/local/src/hw/crochet/work/_.buildkernel.i386-ALIX.log)
Creating a 3900MB raw disk image in:
    /usr/local/src/hw/crochet/work/FreeBSD-i386-11-ALIX-Alix.img
Partitioning the raw disk image with MBR at Mon Feb 26 16:03:18 AKST 2018
gpart create -s MBR md0
md0 created
Creating an auto-sized UFS partition at Mon Feb 26 16:03:18 AKST 2018
md0s1 created
/dev/md0s1a: 3718.4MB (7615360 sectors) block size 32768, fragment size 4096
        using 6 cylinder groups of 626.09MB, 20035 blks, 80256 inodes.
super-block backups (for fsck_ffs -b #) at:
 192, 1282432, 2564672, 3846912, 5129152, 6411392
tunefs: soft updates set
Using inode 4 in cg 0 for 4194304 byte journal
tunefs: soft updates journaling set
tunefs: NFSv4 ACLs set
Installing bootblocks
Boot files are at: /usr/local/src/hw/crochet/work/obj/i386.i386/usr/src/stand/i386
 gpart bootcode -b /usr/local/src/hw/crochet/work/obj/i386.i386/usr/src/stand/i386/mbr/mbr md0
gpart: /usr/local/src/hw/crochet/work/obj/i386.i386/usr/src/stand/i386/mbr/mbr: No such file or directory
    31378.50 real     58297.27 user      3984.73 sys
```

jackavenger, 2020-03-06:
The same on GenericI386:

root@fbsd:~/crochet # ./crochet.sh -c vmware.config.sh
Starting at Fri Mar 6 07:01:20 UTC 2020
Loading configuration from vmware.config.sh
Board: GenericI386
Option: ImageSize 8G
Source version is: 358684
Building FreeBSD version: 12.1
Image name is:
    /root/crochet/work/FreeBSD-i386-12.1-GENERIC-358684-GenericI386.img
Building FreeBSD version: 12.1
Object files are at: /root/crochet/work/obj/usr/src
Found suitable FreeBSD source tree in:
    /usr/src
Using FreeBSD i386 world from previous build
Using FreeBSD i386-GENERIC kernel from previous build
Creating a 8000MB raw disk image in:
    /root/crochet/work/FreeBSD-i386-12.1-GENERIC-358684-GenericI386.img
Partitioning the raw disk image with MBR at Fri Mar 6 07:03:36 UTC 2020
gpart create -s MBR md1
md1 created
Creating an auto-sized UFS partition at Fri Mar 6 07:03:36 UTC 2020
md1s1 created
/dev/md1s1a: 7628.5MB (15623168 sectors) block size 32768, fragment size 4096
        using 13 cylinder groups of 626.09MB, 20035 blks, 80256 inodes.
super-block backups (for fsck_ffs -b #) at:
 192, 1282432, 2564672, 3846912, 5129152, 6411392, 7693632, 8975872, 10258112, 11540352, 12822592, 14104832, 15387072
tunefs: soft updates set
Using inode 4 in cg 0 for 4194304 byte journal
tunefs: soft updates journaling set
tunefs: NFSv4 ACLs set
Installing bootblocks
Boot files are at: /root/crochet/work/obj/usr/src/stand/i386
gpart: /root/crochet/work/obj/usr/src/stand/i386/mbr/mbr: No such file or directory

I found mbr here: /root/crochet/work/obj/usr/src/**i386.i386**/stand/i386/mbr/mbr