freebsd/crochet #173: Build breaks at MK_MAN

adamehirsch, 2017-01-16:
I'm trying to build a freebsd releng/10.3 (from https://svn.freebsd.org/base/releng/10.3) image for a Raspberry Pi 2, using a current crochet at commit `74562852368c3a07c60aeb71e3fbf997a1c5e08b`. The error:

```
root@raven /u/h/a/k/crochet# nice -n 20 ./crochet.sh -c config.sh                                                                                             [328/1941]
Starting at Sun Jan 15 15:06:05 CST 2017
Loading configuration from config.sh
Board: RaspberryPi2
Option: ImageSize 8000mb
Source version is: 312210M
Building FreeBSD version: 10.3
Image name is:
    /usr/home/adam/knox/crochet/work/FreeBSD-armv6-10.3-RPI-B-ZFS-312210M.img
Building FreeBSD version: 10.3
Object files are at: /usr/home/adam/knox/crochet/work/obj/arm.armv6/usr/home/adam/knox/src
Found suitable FreeBSD source tree in:
    /usr/home/adam/knox/src
Found U-Boot port in:
    /usr/local/share/u-boot/u-boot-rpi2
Building FreeBSD armv6 world at Sun Jan 15 15:06:12 CST 2017
    (Logging to /usr/home/adam/knox/crochet/work/_.buildworld.armv6.log)

Building FreeBSD armv6-RPI-B-ZFS kernel at Sun Jan 15 16:40:21 CST 2017
    (Logging to /usr/home/adam/knox/crochet/work/_.buildkernel.armv6-RPI-B-ZFS.log)
Building FreeBSD armv6-RPI-B-ZFS ubldr at Sun Jan 15 16:48:17 CST 2017
    (Logging to /usr/home/adam/knox/crochet/work/ubldr-armv6-RPI-B-ZFS/_.ubldr.armv6-RPI-B-ZFS.build.log)
make: "/usr/home/adam/knox/src/share/mk/bsd.own.mk" line 487: MK_MAN can't be set by a user.
```

The relevant crochet scripts seem to set `MK_MAN=no`:

```
adam@raven ~/k/crochet (master *) [141]> fgrep MK_MAN lib/freebsd.sh lib/i386.sh
lib/freebsd.sh:        eval $buildenv make "$@" DESTDIR=${UBLDR_DIR}/ BINDIR=boot MK_MAN=no -m $ubldr_makefiles install >> ${LOGFILE} || exit 1
lib/freebsd.sh:        eval $buildenv make "$@" DESTDIR=${EFI_DIR}/ BINDIR=boot MK_MAN=no -m $sharemk install >> ${LOGFILE} || exit 1
lib/i386.sh:    if eval ${buildenv} make DESTDIR=${WORKDIR} MK_MAN=no install > ${WORKDIR}/_.i386_loader_install.log 2>&1
```

Any thoughts?

kientzle, 2017-01-16:
Seems maybe we should be using `WITHOUT_MAN=1` instead of `MK_MAN=no`.  Please try that and let me know if that fixes it for you.

adamehirsch, 2017-01-16:
Yup, that worked perfectly.

closed 2017-01-16 by adamehirsch