freebsd/crochet #181: cp: /usr/local/share/u-boot/u-boot-rpi3/armstub8.bin: No such file or directory

erikj, 2017-02-05:
Greetings-

Attempting to build an image for RaspberryPi3 results in the following error:

```sh
cp: /usr/local/share/u-boot/u-boot-rpi3/armstub8.bin: No such file or directory
```

I am attempting build via

```sh
sudo /bin/sh crochet.sh -b RaspberryPi3
```

It looks like the error is produced by code from 43dd6a45, which was recently merged into `master`.

I have installed `aarch64-binutils` and `sysutils/u-boot-rpi3` packages and the `u-boot-rpi3` directory has the following contents:

```sh
$ ls -l /usr/local/share/u-boot/u-boot-rpi3/
total 8835
-rw-r--r--  1 root  wheel     1494 Jan 12 23:16 LICENCE.broadcom
-rw-r--r--  1 root  wheel      123 Jan 12 23:16 README
-rw-r--r--  1 root  wheel    17932 Jan 12 23:16 bootcode.bin
-rw-r--r--  1 root  wheel      136 Jan 12 23:16 config.txt
-rw-r--r--  1 root  wheel     6482 Jan 12 23:16 fixup.dat
-rw-r--r--  1 root  wheel     2504 Jan 12 23:16 fixup_cd.dat
-rw-r--r--  1 root  wheel     9722 Jan 12 23:16 fixup_db.dat
-rw-r--r--  1 root  wheel     9716 Jan 12 23:16 fixup_x.dat
-rw-r--r--  1 root  wheel  2747896 Jan 12 23:16 start.elf
-rw-r--r--  1 root  wheel   618296 Jan 12 23:16 start_cd.elf
-rw-r--r--  1 root  wheel  4928152 Jan 12 23:16 start_db.elf
-rw-r--r--  1 root  wheel  3879416 Jan 12 23:16 start_x.elf
-rw-r--r--  1 root  wheel   354344 Jan 12 23:16 u-boot.bin
```

I am relatively new to FreeBSD and this is my first attempt @ building an image w/ `crochet`, so if I am doing something wrong and you can provide guidance, it would be appreciated.

Thank you for supporting this cool project :)

brd, 2017-02-05:
Your u-boot-rpi3 is too old, the packages have not been updated yet.  What version of FreeBSD are you running?

erikj, 2017-02-05:
> What version of FreeBSD are you running?

Thanks.

11.0-RELEASE-p2

kientzle, 2017-02-12:
Have you tried updating the `sysutils/u-boot-rpi3` package recently?  I think that file should be present in the pre-built packages by now.

rozmansi, 2017-02-24:
The u-boot-rpi3 in the FreeBSD-11.0 pkg repository is version _2016.11_.
The u-boot-rpi3 in the ports collection is version _2017.01_.

The later indeed contains all required files to complete a building of RPi3 image. :)

Unfortunately, it does not boot. Probably, because the boot partition is FAT12 >> filenames longer than 8.3 get truncated. But that's another issue.

erikj, 2017-02-24:
@kientzle @rozmansi Thank you, I will give the build another try.

rozmansi, 2017-02-24:
I fixed the 8+3 FAT filename problem by modifying lib/disk.sh:
   mount_msdosfs `disk_fat_device $2` $1
to
   mount_msdosfs -l `disk_fat_device $2` $1

rozmansi, 2017-05-22:
I've made a commit be549bbfc1953dfd7bbfeaba3e24fc868cd403bd which was pulled in. So this issue should be closed as "resolved".

closed 2017-07-09 by kientzle