freebsd/crochet #182: Problems with NanoBSD config for RPI3

tickerguy, 2017-02-11:
The partition creation in NanoBSD/setup.sh is defective.  Specifically, the fstab file is built with the assumption that cfg is on "d" and data is on "e", but gpart is not called with an index to create those two partitions, which means they get "next slice" which aligns wrong.

In addition /boot/msdos does not exist as a directory on all distributions and it shouldn't be mounted all the time anyway (only when changes are needed).

As published the system builds but on boot it fails due to mount and partition name errors on the Pi2 and 3 (and probably others, since the same issue will arise with the partition index values.)  Note that while the instructions say that NanoBSD can produce two system partitions in practice it does not; that is not addressed.

The below addresses both of the above issues for a single-system partition build:

root@NewFS:/pics/Crochet/option/NanoBSD # diff -u /pics/Crochet-base/option/NanoBSD/setup.sh setup.sh
--- /pics/Crochet-base/option/NanoBSD/setup.sh  2017-02-11 16:04:19.564467000 -0600
+++ setup.sh    2017-02-11 15:43:57.419217000 -0600
@@ -54,13 +54,13 @@

     # CFG Paritition
     #CFG_UFS_PARTITION=`gpart add -t freebsd-ufs ${NANO_CFG_SIZE} -l cfg ${NEW_UFS_SLICE} | sed -e 's/ .*//'` || exit 1
-    CFG_UFS_PARTITION=`gpart add -t freebsd-ufs ${NANO_CFG_SIZE} ${NEW_UFS_SLICE} | sed -e 's/ .*//'` || exit 1
+    CFG_UFS_PARTITION=`gpart add -t freebsd-ufs -i 4 ${NANO_CFG_SIZE} ${NEW_UFS_SLICE} | sed -e 's/ .*//'` || exit 1
     CFG_UFS_DEVICE=/dev/${CFG_UFS_PARTITION}
     newfs ${CFG_UFS_DEVICE}

     # DATA Paritition
     #DATA_UFS_PARTITION=`gpart add -t freebsd-ufs -l data ${NEW_UFS_SLICE} | sed -e 's/ .*//'` || exit 1
-    DATA_UFS_PARTITION=`gpart add -t freebsd-ufs ${NEW_UFS_SLICE} | sed -e 's/ .*//'` || exit 1
+    DATA_UFS_PARTITION=`gpart add -t freebsd-ufs -i 5 ${NEW_UFS_SLICE} | sed -e 's/ .*//'` || exit 1
     DATA_UFS_DEVICE=/dev/${DATA_UFS_PARTITION}
     newfs ${DATA_UFS_DEVICE}
     tunefs -n enable ${DATA_UFS_DEVICE}                        # Turn on Softupdates
@@ -96,7 +96,7 @@

     # add /cfg and /data to /etc/fstab
     cp /dev/null etc/fstab
-    echo "${NANO_DEV}s1  /boot/msdos  msdosfs  ro                 1 1" >> etc/fstab
+    echo "${NANO_DEV}s1  /boot/msdos  msdosfs  ro,noauto          1 1" >> etc/fstab
     echo "${NANO_DEV}s2a /            ufs      ro                 1 1" >> etc/fstab
     echo "${NANO_DEV}s2d /cfg         ufs      rw,noatime,noauto  2 2" >> etc/fstab
     echo "${NANO_DEV}s2e /data        ufs      rw,noatime         2 2" >> etc/fstab

kientzle, 2017-02-12:
Please look at e1d7dc32 and verify that I've applied the index fix correctly.

I've not applied your change to `/boot/msdos` because I'm confused about something:  
It seems to me that if `/boot/msdos` isn't actually present everywhere this option is used, then we shouldn't be adding that entry to `/etc/fstab` in this option.  If that's specific to only certain boards, then the board definition should be updating `/etc/fstab` as necessary.

tickerguy, 2017-02-12:
Looks ok.

The other change, however (/boot/msdos) is one that either has to be made or the entry has to be pulled entirely from the NanoBSD setup (and left to the board definition), because as it stands now if that directory is missing the build succeeds but won't come up multiuser, since the fstab entry refers to a directory that doesn't exist.

Removing /boot/msdos entirely from the NanoBSD config is arguably the right choice as it certainly isn't universally true that such a partition would be present in all configurations.

kientzle, 2017-02-13:
I agree.  I've removed the `/boot/msdos` entry here in commit 4bef01795de

Thanks!

closed 2017-02-13 by kientzle