freebsd/crochet #183: NanoBSD setup.sh fails to create copy and link for usr/local/etc

tickerguy, 2017-02-14:
Synopsis: /usr/local/etc isn't in the default system; it gets created when you install packages.  The Crochet build process applies user entries at the **END** of the process, which means that /usr/local/etc does not exist at the time NanoBSD/setup.sh runs.  This in turn means that the copy never occurs unless you have some other policy matter that gets done before the image build takes place.

There is no harm from having usr/local/etc mirrored in the config directory but plenty of harm from not if you have the customization of the build happen in the customize_freebsd_partition() call then it's too late because NanoBSD/setup.sh has already run.  This means no symlink, which in turn means all your local "stuff" is missing its config.

Since there's no harm or foul in creating this link in any event and it **has** to be there if you intend to store anything (like configuration files!) in /usr/local/etc, this patch changes how that functions to:

1. Copy the local directory over to conf if it already exists when NanoBSD runs
2. But, if it doesn't, make the symlink anyway since the "local" directory in etc is already there in a Nano build, thus the link is valid -- and if populated later it's ok.

Note that the user of same remains responsible for making sure the conf/base/etc/local/..... files wind up there.  If they don't, go in usr/local/etc instead they won't be there once the system finishes booting.

```--- ../../../Crochet-base/option/NanoBSD/setup.sh       2017-02-14 14:10:51.765528000 -0600
+++ setup.sh    2017-02-14 12:45:22.046988000 -0600
@@ -80,13 +80,16 @@
         (
         cd usr/local/etc
         find . -print | cpio -dumpl -R root:wheel ../../../etc/local
         cd ..
         rm -fr ./etc
-        ln -s ../../etc/local ./etc
         )
     fi
+    (
+    cd usr/local
+    ln -s ../../etc/local etc
+    )

     # force diskless mode
     touch etc/diskless
     chown root:wheel etc/diskless
```

amanharitsh123, 2017-09-21:
Hi, i am new to open source ..i just want to give a try to fix this issue..Can you please guide me a bit

kientzle, 2017-09-23:
@amanharitsh123:  It's easy:

Step 1:  Click the "fork" button in the upper right of the Github website.  That will create a copy of this project in your account.

Step 2: Check out your copy of this project, make the fixes and "git push" the fixed version into your copy.

Step 3: Create a "Pull Request" that asks us to pull the changes from your copy into our copy.  We'll take a look at your changes and discuss it with you.

Of course, this works best if you are sending us only a few specific changes at a time; one common reason for refusing a pull request is when it has a lot of unrelated changes that make it hard to understand.  For this reason, experienced developers create git branches in their copy with each branch holding only a specific focused set of changes.  If you're unfamiliar with git, there are many good books and web sites that talk about how to do this.