freebsd/crochet #186: Beaglebone Black not booting: No valid device tree blob found!

bsdhsr, 2017-02-27:
Tried downloading the most recent snapshot from the ftp server, was greeted with "No valid device tree blob found!"

Built a custom image with crochet, but the error remains.

11-STABLE

---

U-Boot SPL 2017.01-rc3 (Feb 26 2017 - 03:19:59)
Trying to boot from MMC1MMC partition switch failed
*** Warning - MMC partition switch failed, using default environment

reading u-boot.img
reading u-boot.img

U-Boot 2017.01-rc3 (Feb 26 2017 - 03:19:59 +0000)

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
*** Warning - bad CRC, using default environment

<ethaddr> not set. Validating first E-fuse MAC
Net:   cpsw, usb_ether
Press SPACE to abort autoboot in 2 seconds
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
reading boot.scr
** Unable to read file boot.scr **
reading uEnv.txt
22 bytes read in 4 ms (4.9 KiB/s)
Loaded env from uEnv.txt
Importing environment from mmc0 ...
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found FreeBSD U-Boot Loader (bin)
reading ubldr.bin
231000 bytes read in 21 ms (10.5 MiB/s)
## Starting application at 0x82000000 ...
Consoles: U-Boot console  
Compatible U-Boot API signature found @0x9df30c58

FreeBSD/armv6 U-Boot loader, Revision 1.2
(Mon Feb 27 19:17:27 UTC 2017 root@localhost)

DRAM: 512MB
Number of U-Boot devices: 3
U-Boot env: loaderdev not set, will probe all devices.
Found U-Boot device: disk
  Probing all disk devices...
  Checking unit=0 slice=<auto> partition=<auto>... good.
Booting from disk0s2a:
/boot/kernel/kernel data=0x6d4a64+0x14759c syms=[0x4+0x7f5c0+0x4+0x92645]

Hit [Enter] to boot immediately, or any other key for command prompt.
Booting [/boot/kernel/kernel]...               
No valid device tree blob found!

Type '?' for a list of commands, 'help' for more detailed help.
loader>

prasadboraskar, 2017-07-05:
I am seeing the same issue. Any update on the problem?

prasadboraskar, 2017-07-06:
Looks like its the snapshot that was the problem. The older builds had some changes which caused this fatal kernel issue. I used the latest snapshot "FreeBSD-12.0-CURRENT-arm-armv6-BEAGLEBONE-20170626-r320360” and it seems to work fine. 

Heres the link which explains it a bit more and the comment from the person who fixed it.
http://freebsd.1045724.x6.nabble.com/Fatal-kernel-abort-was-r313878M-builds-mmcsd0s2-as-read-only-amp-etc-fstab-is-missing-td6169063.html

**_Yep, it turned out I caused this with my changes to support gpio pins 
for card detect and write protect.  The fdt data for beaglebone doesn't 
list any pins for write protect and doesn't have the "wp-disable" 
property (which is a bug in the fdt data, because eMMC and micro-sd 
cannot be write protected).  The old driver logic treated that as "not 
protected" and a side effect of the way the new code works effectively 
reversed that into saying everything is write protected. 

I just commited a fix for it as r314071, sorry for taking so long to 
realize all of this was fallout from my change.**_

kientzle, 2017-07-09:
Sounds like this is a FreeBSD problem that has been fixed, and not a problem with Crochet itself.

closed 2017-07-09 by kientzle