freebsd/crochet #187: Build failure: cd: /tank/rpi3/crochet/obj/arm64.aarch64/tank/rpi3/src/sys/AEGIR: No such file or directory

gczuczy, 2017-03-08:
I'm trying to an RPi3 image from -CURRENT, since the raspbsd image doesn't have device spigen enabled. I'm getting the following error at buildkernel:

I've attached the buildkernel log.

The kernel config is symlinked to the source tree:
f# ls -lh /tank/rpi3/src/sys/arm64/conf/AEGIR
lrwxr-xr-x  1 root  wheel    19B Mar  8 06:56 /tank/rpi3/src/sys/arm64/conf/AEGIR -> /etc/kernconf/AEGIR

It's a simple include GENERIC, ident AGEIR and device spigen.

I've attached the crochet config file as aegir.txt.

Could you please let me know what am I messing up? Or if it isn't me, how can this be corrected?

[aegir.txt](https://github.com/freebsd/crochet/files/827011/aegir.txt)

[_.buildkernel.aarch64-AEGIR.log.txt](https://github.com/freebsd/crochet/files/827005/_.buildkernel.aarch64-AEGIR.log.txt)

gczuczy, 2017-03-08:
I think this was due to accidentally enabling KERNFAST. After removing it, kernel is successfully built.

kientzle, 2017-03-11:
Glad you got it working.

closed 2017-03-11 by kientzle