freebsd/crochet #188: Make it possible to specify image directory

gczuczy, 2017-03-09:
Currently it's not possible to use the automatic naming with the checkout revision and everything, into a custom directory. When using -CURRENT (like for RPis it's common) it would be nice to retain the default naming convention, however just using a separate output directory.

Currently the code in lib/board.sh looks like:
```
board_generate_image_name ( ) {
    if [ -z "${IMG}" ]; then
        if [ -z "${SOURCE_VERSION}" ]; then
           IMG=${WORKDIR}/FreeBSD-${TARGET_ARCH}-${FREEBSD_MAJOR_VERSION}-${KERNCONF}.img
       else
           IMG=${WORKDIR}/FreeBSD-${TARGET_ARCH}-${FREEBSD_VERSION}-${KERNCONF}-${SOURCE_VERSION}.img
       fi
    fi
```

Personally, I would suggest adding an IMGDIR variable, defaulting to WORKDIR, this way we could simply redirect the output to a different directory, without having to mess around with the naming. Manually specifying the same naming scheme doesn't work, because those variables are not available at the time the configuration file is being interpreted.

kientzle, 2017-07-09:
Could you please send a pull request with your suggested change?

gczuczy, 2017-07-10:
Sure, I just have to find the time, when I can look into it.

gczuczy, 2017-08-17:
I've added support for IMGDIR and IMGNAME.

If IMGDIR is not specified, then WORKDIR will be used.

if IMGNAME is specified, then it's evaluated, instead of the defaults. This way the same variables can still be used, which are not available when the configuration file is being run.

Here are some test cases:

1) Only IMGDIR is speified:
```
IMGDIR=/tank/rpi3/images/
```
Result:
```
Image name is:
    /tank/rpi3/images//FreeBSD-aarch64-12.0-GENERIC-322617.img
```

Only IMGNAME is specified (notice the single apostrophes):
```
IMGNAME='FreeBSD-${FREEBSD_VERSION}-foo-${KERNCONF}-bar-${SOURCE_VERSION}'
```
Result:
```
Image name is:
    /tank/rpi3/crochet/work/FreeBSD-12.0-foo-GENERIC-bar-322617
```
It still defaults to ${TOPDIR}/work

And when both IMGDIR and IMGNAME are specified:
```
IMGDIR=/tank/rpi3/images/
IMGNAME='FreeBSD-${FREEBSD_VERSION}-foo-${KERNCONF}-bar-${SOURCE_VERSION}'
```
Result:
```
Image name is:
    /tank/rpi3/images//FreeBSD-12.0-foo-GENERIC-bar-322617
```

I will add this to the config.sh.sample, so people will know about it. After I'm done with that, will submit the pull request.

gczuczy, 2017-08-17:
@kientzle Here it is: https://github.com/freebsd/crochet/pull/206

kientzle, 2017-08-18:
Thanks!  Merged.

closed 2017-08-18 by kientzle

gczuczy, 2017-08-20:
@kientzle  You are welcome.