freebsd/crochet #190: wrong u-boot filename for banana-pi if installed from ports

MoritzFago, 2017-04-11:
I tried to create a banana-pi image with crochet and installed the u-boot for banana-pi from the ports tree. The setup.sh looks for `sunxi-spl.bin`, but there is only a `u-boot-sunxi-with-spl.bin` file.

kientzle, 2017-07-09:
Can you please submit a Pull Request with the fix?  Thanks.

kientzle, 2017-07-29:
I believe this is fixed by #199.

closed 2017-07-29 by kientzle

MoritzFago, 2017-07-29:
wouldn't it be easier to understand/debug if you would specify the u-boot file name somewhere depending on what was found, log that and then use dd once? As far as i can see, in the merged pull request #199 you use dd twice, what could case the boot loader to be written twice, if for example the package and the port are installed.

kientzle, 2017-07-29:
> ...  if you would specify the u-boot file name somewhere depending on what was found, log that and then use...

Could you send us a pull request showing exactly how you think that could be implemented?

> ... you use dd twice ...

I'm not sure I understand.  Are you talking about these two lines?
```
dd if=${UBOOT_PATH}/u-boot-sunxi-with-spl.bin conv=notrunc,sync of=/dev/${DISK_MD} bs=1024 seek=8
dd if=${UBOOT_PATH}/u-boot.img conv=notrunc,sync of=/dev/${DISK_MD} bs=1024 seek=40
```
Those are installing different things to different places in the image.

PR #199 (2017-07-29): rename sunxi-spl.bin to u-boot-sunxi-with-spl.bin on all Allwinner boards
As the result of [r431327](https://svnweb.freebsd.org/ports?view=revision&revision=431327) in ports.
Also use conv=notrunc,sync when writing, as mentioned in https://github.com/freebsd/crochet/issues/191#issuecomment-313300456.

Fixes #190 
Fixes #191
commit 2c99e9ce (2017-07-29): rename sunxi-spl.bin to u-boot-sunxi-with-spl.bin on all Allwinner boards

As the result of https://svnweb.freebsd.org/ports?view=revision&revision=431327.
Also use conv=notrunc,sync when writing, as mentioned in
https://github.com/freebsd/crochet/issues/191#issuecomment-313300456.

Fixes #190, #191.