freebsd/crochet #191: dd: /dev/md4: Invalid argument on u-boot installation for banana-pi

MoritzFago, 2017-04-11:
During the image creation process the setup.sh burns the various boot loader files to the image.
```csh
Installing U-Boot files
+ dd if=/usr/local/share/u-boot/u-boot-bananapi/sunxi-spl.bin of=/dev/md4 bs=1024 seek=8
dd: /dev/md4: Invalid argument
486+1 records in
486+0 records out
497664 bytes transferred in 5.784579 secs (86033 bytes/sec)
```
I am building in a jail with the following flags `allow.mount=true,enforce_statfs=1`

fjerhammer, 2017-04-22:
This is also happening for me. 11.0-STABLE, not using jails.

arrowd, 2017-05-03:
And me too.

alexey1234, 2017-05-28:
And me too

swilde, 2017-06-02:
Here too.

arrowd, 2017-06-18:
So, is crochet completely unusable right now? The offending line is https://github.com/freebsd/crochet/blob/master/lib/disk.sh#L47 but I don't know what to do with it.

arrowd, 2017-06-18:
Ah, no, that line has nothing to do with the problem.

kientzle, 2017-07-09:
To debug this, you can (temporarily) edit `board/BananaPi/setup.sh` to print out more information at the top of the `allwinner_partition_image()` function.  In particular, it would probably make sense to try `ls -l /dev/${DISK_MD}` to make sure the md device was created, and maybe `mdconfig -l` to list all the existing md devices.

kientzle, 2017-07-09:
@loos-br might be right.  If /usr/local/share/u-boot/u-boot-bananapi/sunxi-spl.bin is not a multiple of the block size, then adding `conv=sync` to the `dd` command originally mentioned may fix it.

kientzle, 2017-07-29:
I believe #199 fixes this; please try and let us know.

PR #199 (2017-07-29): rename sunxi-spl.bin to u-boot-sunxi-with-spl.bin on all Allwinner boards
As the result of [r431327](https://svnweb.freebsd.org/ports?view=revision&revision=431327) in ports.
Also use conv=notrunc,sync when writing, as mentioned in https://github.com/freebsd/crochet/issues/191#issuecomment-313300456.

Fixes #190 
Fixes #191
closed 2017-07-29 by kientzle