freebsd/crochet #192: Support for Raspberry Pi Zero W

JeanVEGA, 2017-04-15:
Hi,

Could you please add https://www.raspberrypi.org/blog/raspberry-pi-zero-w-joins-family/ build support? I am running to this issue https://www.raspberrypi.org/forums/viewtopic.php?f=85&t=179244&p=1143307 with RPI-B build.

Attempted to replace latest firmware https://lists.freebsd.org/pipermail/freebsd-arm/2015-December/012759.html from https://github.com/raspberrypi/firmware, I only got to the "rainbow screen" https://www.raspberrypi.org/forums/viewtopic.php?f=31&t=37068.

Thanks!

kientzle, 2017-07-09:
The freebsd-arm mailing list is a much better place to ask about FreeBSD support for new boards.

  https://www.freebsd.org/community/mailinglists.html

closed 2017-07-09 by kientzle

JeanVEGA, 2017-07-10:
Posted a question https://forums.freebsd.org/threads/60610/. AFAIK WiFi support on FreeBSD ARM is blocked by lack of https://wiki.freebsd.org/SDIO implementation. Discussed https://lists.freebsd.org/pipermail/freebsd-arm/2016-February/013286.html and review https://reviews.freebsd.org/D4761, inactive for more than a year now.