freebsd/crochet #194: FreeBSD 11 - Pine64 Build Problem

okeozkilic, 2017-05-11:
Hi,

I must say that I am not really a competent "compile" person and I am stuck at building a Pine64 image.  My config file options are very simple and I did really go through all the steps. When I try to build the image I get this error some (about 15 minutes) time after I start building (below).

I am using FreeBSD 11.0 arm64 (patch 8) on VirtualBox. Tried FreeBSD 11.0 i386 and got the same errors all the time. I cannot even say what is wrong, what the error message says, which file is really the problem. Any help on this topic will be truly appreciated. 

I was able to build the image (same config file, just changed the board comment) for RPI2 without any problems. 

--- libc.so.7.full ---
/usr/local/aarch64-freebsd/bin/ld: getutxent.So(.debug_info+0x3c): R_AARCH64_ABS64 used with TLS symbol udb
/usr/local/aarch64-freebsd/bin/ld: getutxent.So(.debug_info+0x59): R_AARCH64_ABS64 used with TLS symbol uf
/usr/local/aarch64-freebsd/bin/ld: utxdb.So(.debug_info+0x5c): R_AARCH64_ABS64 used with TLS symbol futx_to_utx.ut
/usr/local/aarch64-freebsd/bin/ld: jemalloc_tsd.So(.debug_info+0x39): R_AARCH64_ABS64 used with TLS symbol __je_tsd_tls
/usr/local/aarch64-freebsd/bin/ld: jemalloc_tsd.So(.debug_info+0x1430): R_AARCH64_ABS64 used with TLS symbol __je_tsd_initialized
/usr/local/aarch64-freebsd/bin/ld: cxa_thread_atexit.So(.debug_info+0x3c): R_AARCH64_ABS64 used with TLS symbol dtors
/usr/local/aarch64-freebsd/bin/ld: xlocale.So(.debug_info+0x404): R_AARCH64_ABS64 used with TLS symbol __thread_locale
/usr/local/aarch64-freebsd/bin/ld: setrunelocale.So(.debug_info+0x3d): R_AARCH64_ABS64 used with TLS symbol _ThreadRuneLocale
cc: error: linker command failed with exit code 1 (use -v to see invocation)
*** [libc.so.7.full] Error code 1

make[4]: stopped in /usr/src/lib/libc
1 error

make[4]: stopped in /usr/src/lib/libc
*** [lib/libc__L] Error code 2

make[3]: stopped in /usr/src
1 error

make[3]: stopped in /usr/src
*** [libraries] Error code 2

make[2]: stopped in /usr/src
1 error

make[2]: stopped in /usr/src
*** [_libraries] Error code 2

make[1]: stopped in /usr/src
1 error

make[1]: stopped in /usr/src
*** [buildworld] Error code 2

make: stopped in /usr/src
1 error

make: stopped in /usr/src

rozmansi, 2017-05-21:
It's a problem with aarch64 (arm64). I had the same problem compiling for Raspberry Pi 3.

Then I found [this memo](http://freebsd.1045724.x6.nabble.com/I-had-to-revert-usr-local-aarch64-freebsd-from-2-28-for-its-bin-ld-to-work-for-r315870-buildworld-ad-td6175882.html).

1. You need to switch from `portsnap` to `svn` ports tree.
2. Go to `/usr/ports` and do the selective revert:
```
svnlite update -r436731 devel/binutils \
                          devel/aarch64 \
                          devel/aarch64-none-elf-binutils \
                          devel/arm-gnueabi-binutils \
                          devel/arm-none-eabi-binutil
```
3. Rebuild and reinstall those ports.
4. Clean and rebuild your Pine64 image using Crochet.

rozmansi, 2017-05-22:
As this issue is not related to Crochet itself, I suggest it is closed.

closed 2017-05-22 by brd