freebsd/crochet #200: cubieboard2 11.1-RELEASE build fails to boot with "No valid device tree blob found!"

paulzhol, 2017-07-29:
drops into `loader>` prompt. Looks exactly like #186. I think it's caused by the new u-boot-sunxi-with-spl.bin 2017.01 u-boot.

I've ended up with this in my /boot/loader.rc.local:
```
load /boot/kernel/kernel
load -t dtb /boot/dtb/cubieboard2.dtb
```
Should this file be added to the cubieboard2 (and probably the rest of the allwinner boards) overlay?

kientzle, 2017-07-29:
You should ask on the freebsd-arm mailing list and see what those folks recommend.

More likely, recent changes to the U-Boot port have changed the expected DTB file name.  I know Warner and other FreeBSD folks have been trying to make the various U-Boot ports more consistent and some of the file names have been changing around in the process.

zilti, 2017-07-30:
Same with my Orange Pi Plus 2, just that I have to do `load -t dtb /boot/dtb/orangepi-plus-2e.dtb` instead.

paulzhol, 2017-07-30:
I've looked at the official image (FreeBSD-11.1-RELEASE-arm-armv6-CUBIEBOARD2.img.xz).
Attached the console [log](https://gist.github.com/paulzhol/9f341d89335f20fb646fc6ae141f4bb4). It is loading `/boot/dtb/sun7i-a20-cubieboard2.dtb` and proceeding correctly.

I've removed my  `/boot/loader.rc.local` and hard-linked  `/boot/dtb/sun7i-a20-cubieboard2.dtb`  to `/boot/dtb/cubieboard2.dtb`. It's booting correctly!

I think the dtb name is coming from u-boot, Debian wiki has a list of supported boards and their respective dtbs [here](https://wiki.debian.org/InstallingDebianOn/Allwinner#Supported_Platforms).