freebsd/crochet #201: pine64 build no longer works (since u-boot-pine64-2017.07.00.1)

hoopty, 2017-08-03:
After updating u-boot-pine64 to 2017.07.00.1, there is no more pine64.img that the Pine64 board  setup.sh is using. @evadot, if you have advice on installing the U-boot instead of the following, I'll test and incorporate into a PR:
```
dd if=${PINE64_UBOOT_PATH}/${PINE64_UBOOT_BIN} of=/dev/${DISK_MD} seek=16
```

bsdimp, 2017-08-04:
PINE64_UBOOT_BIN=u-boot-sunxi-with-spl.bin
somewhere should fix this.

This is why, btw, I keep harping on the need to put this kind of metadata into the u-boot ports and having a common script to create these images rather than playing whack-a-mole with all the image creation scripts.

kientzle, 2017-08-05:
> ...need to put this kind of metadata into the u-boot ports...

For boards that just copy the U-Boot files onto a FAT partition, "this kind of metadata" would just be the list of filenames.  But for boards that require the various U-Boot bits to be copied into the image at particular byte offsets, it's much more complex.

Putting this data entirely into the U-Boot ports basically means building an API between "all the image creation scripts" and all the U-Boot ports.

closed 2017-08-07 by hoopty