freebsd/crochet #203: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__

argot42, 2017-08-07:
I'm trying to build an image for a raspi b+ and I got this error in the process.

```
    % sudo /bin/sh crochet.sh -c config.sh
    Starting at Wed Jul 5 16:36:37 ART 2017
    Loading configuration from config.sh.sample
    Board: RaspberryPi
    Option: ImageSize 3900mb
    Option: User argot
    Option: SwapFile 1gb [deferred] [file=/swapfile0]
    Option: UsrPorts
        Will install up-to-date /usr/ports from portsnap
    Option: Ntpd
    Source version is: 320667
    Building FreeBSD version: 12.0
    Image name is:
        /usr/home/argot/crochet/work/FreeBSD-RPI-B.img
    Building FreeBSD version: 12.0
    Object files are at: /usr/home/argot/crochet/work/obj/arm.armv6/usr/src
    Found suitable FreeBSD source tree in:
        /usr/src
    Found U-Boot port in:
        /usr/local/share/u-boot/u-boot-rpi
    Building FreeBSD armv6 world at Wed Jul 5 16:36:50 ART 2017
        (Logging to /usr/home/argot/crochet/work/_.buildworld.armv6.log)
    Building FreeBSD armv6-RPI-B kernel at Wed Jul 5 22:03:38 ART 2017
        (Logging to /usr/home/argot/crochet/work/_.buildkernel.armv6-RPI-B.log)
    Building FreeBSD armv6-RPI-B ubldr at Wed Jul 5 22:23:17 ART 2017
        (Logging to /usr/home/argot/crochet/work/ubldr-armv6-RPI-B/_.ubldr.armv6-RPI-B.build.log)
    Creating a 3900MB raw disk image in:
        /usr/home/argot/crochet/work/FreeBSD-RPI-B.img
    Partitioning the raw disk image with MBR at Wed Jul 5 22:23:37 ART 2017
    gpart create -s MBR md0
    md0 created
    Creating a 17m FAT partition at Wed Jul 5 22:23:37 ART 2017 with start block 63 and label BOOT
    active set on md0s1
    Creating an auto-sized UFS partition at Wed Jul 5 22:23:40 ART 2017
    md0s2 created
    /dev/md0s2a: 3701.4MB (7580544 sectors) block size 32768, fragment size 4096
        using 6 cylinder groups of 626.09MB, 20035 blks, 80256 inodes.
    super-block backups (for fsck_ffs -b #) at:
     192, 1282432, 2564672, 3846912, 5129152, 6411392
    tunefs: soft updates set
    Using inode 4 in cg 0 for 4194304 byte journal
    tunefs: soft updates journaling set
    tunefs: NFSv4 ACLs set
    Mounting all file systems:
    Mounting FAT partition 1 at /usr/home/argot/crochet/work/_.mount.boot
    Mounting UFS partition 1 at /usr/home/argot/crochet/work/_.mount.freebsd
    ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
    ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
    ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
```
But ```find / -name "*local_fixups*"``` throws no results.

Here is my [crochet configuration](https://pastebin.com/SwRgNvFF)

Thanks in advance !

kientzle, 2017-08-18:
Can you add some `echo` statements to `board/RaspberryPi/setup.sh` and see if you can figure out exactly which line is triggering these messages?

It might also help to know the exact version of FreeBSD that's running on your build host.

I suspect this is coming from `dtc` when we try to compile the device tree file.  If I'm right, then this is referring to a property in one of the FDT files in the FreeBSD source tree.

bsdimp, 2017-08-18:
These messages are from dtc...

__local_fixups__ is a special property generated entirely by dtc to support overlays. You won't find it in the source tree.

Are you using the in-tree BSDL dtc, or the dtc port?

argot42, 2017-08-31:
@kientzle No matter where I put the `echo` statement in the `setup.sh` file it gets printed at the beginning of the process. Are there other scripts being run after `setup.sh` ?

The exact version of the build host is `11.0-RELEASE-p1`.

@bsdimp I'm using dtc from the in-tree BSDL.

kientzle, 2017-09-05:
> Are there other scripts being run after setup.sh ?

Yes, many.  As the name suggests, `setup.sh` just sets up various parameters (and defines some custom functions for the board).

Here are some `echo` statements that should clarify things.  This will print out the exact `make_dtb.sh` command that's being run and print some messages before and after that command to verify that this is in fact the command that's generating the above messages.

```patch
diff --git a/lib/freebsd.sh b/lib/freebsd.sh
index 38608e9..b6a5f3d 100644
--- a/lib/freebsd.sh
+++ b/lib/freebsd.sh
@@ -512,14 +512,19 @@ freebsd_install_fdt ( ) (
                    exit 1
                    ;;
            esac
+           echo "About to run make_dtb.sh"
+           echo ${FREEBSD_SRC}/sys/tools/fdt/make_dtb.sh ${FREEBSD_SRC}/sys ${_DTSIN} ${_DTBINTERMEDIATE}
            echo ${FREEBSD_SRC}/sys/tools/fdt/make_dtb.sh ${FREEBSD_SRC}/sys ${_DTSIN} ${_DTBINTERMEDIATE} | (cd ${FREEBSD_SRC}; make TARGET_ARCH=$TARGET_ARCH buildenv > /dev/null)
+           echo "Just finished make_dtb.sh"
             case $2 in
                 *.dts)
                    _DTSOUT=$2
+                   echo "Converting dtb to dts"
                    dtc -I dtb -O dts ${_DTBINTERMEDIATEDIR}/*.dtb > ${_DTSOUT}
                     ;;
                 *.dtb)
                    _DTBOUT=$2
+                   echo "Copying dtb"
                    cp ${_DTBINTERMEDIATEDIR}/*.dtb ${_DTBOUT}
                     ;;
                 *)
```

bsdimp, 2017-09-05:
I'm pretty sure this is due to the dtc move from the GPL one to the BSDL one...

argot42, 2017-09-16:
@kientzle I made those changes and this is the result
```Starting at Fri Sep 15 22:43:32 ART 2017
Loading configuration from config.sh
Board: RaspberryPi
Option: ImageSize 3900mb
Option: User argot
Option: SwapFile 1gb [deferred] [file=/swapfile0]
Option: UsrPorts 
    Will install up-to-date /usr/ports from portsnap
Option: Ntpd 
Source version is: 320667
Building FreeBSD version: 12.0
Image name is:
    /usr/home/argot/crochet/work/FreeBSD-RPI-B.img
Building FreeBSD version: 12.0
Object files are at: /usr/home/argot/crochet/work/obj/arm.armv6/usr/src
Found suitable FreeBSD source tree in:
    /usr/src
Found U-Boot port in:
    /usr/local/share/u-boot/u-boot-rpi
Using FreeBSD armv6 world from previous build
Using FreeBSD armv6-RPI-B kernel from previous build
Using ubldr from previous build
Creating a 3900MB raw disk image in:
    /usr/home/argot/crochet/work/FreeBSD-RPI-B.img
Partitioning the raw disk image with MBR at Fri Sep 15 22:43:36 ART 2017
gpart create -s MBR md9
md9 created
Creating a 17m FAT partition at Fri Sep 15 22:43:36 ART 2017 with start block 63 and label BOOT
active set on md9s1
Creating an auto-sized UFS partition at Fri Sep 15 22:43:37 ART 2017
md9s2 created
/dev/md9s2a: 3701.4MB (7580544 sectors) block size 32768, fragment size 4096
	using 6 cylinder groups of 626.09MB, 20035 blks, 80256 inodes.
super-block backups (for fsck_ffs -b #) at:
 192, 1282432, 2564672, 3846912, 5129152, 6411392
tunefs: soft updates set
tunefs: soft updates journaling set
Using inode 4 in cg 0 for 4194304 byte journal
tunefs: NFSv4 ACLs set
Mounting all file systems:
Mounting FAT partition 1 at /usr/home/argot/crochet/work/_.mount.boot
   Removing already-existing mount directory.
   Removed pre-existing mount directory; creating new one.
Mounting UFS partition 1 at /usr/home/argot/crochet/work/_.mount.freebsd
   Removing already-existing mount directory.
   Removed pre-existing mount directory; creating new one.
About to run make_dtb.sh
/usr/src/sys/tools/fdt/make_dtb.sh /usr/src/sys /usr/src/sys/boot/fdt/dts/arm/rpi.dts /usr/home/argot/crochet/work/fdt/fdt.Wz0WuY
Just finished make_dtb.sh
copying dtb
About to run make_dtb.sh
/usr/src/sys/tools/fdt/make_dtb.sh /usr/src/sys /usr/src/sys/boot/fdt/dts/arm/rpi.dts /usr/home/argot/crochet/work/fdt/fdt.rjvNIZ
Just finished make_dtb.sh
Converting dtb to dts
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
[... 155 lines cut ...]
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
[... cut ...]
<stdout>: ERROR (duplicate_property_names): Duplicate property name fixup in /__local_fixups__
ERROR: Input tree has errors, aborting (use -f to force output)
```
it fails exactly on this line ```dtc -I dtb -O dts ${_DTBINTERMEDIATEDIR}/*.dtb > ${_DTSOUT}```

@bsdimp I tried changing the dtc that ships with freebsd with the one you can install with ```pkg install dtc``` but no dice, it fails in the same line. (I don't know if that's the correct way to do it, it's the first time using a BSD system).

kientzle, 2017-09-30:
`dtc -I dtb -O dts ${_DTBINTERMEDIATEDIR}/*.dtb > ${_DTSOUT}`

This line is building a 'dts' input file from the compiled 'dtb'.  The 'dts' file is not technically necessary and is only being built here for reference purposes.  You could comment out this line and it would probably work correctly.

argot42, 2017-10-07:
It worked, many thanks!

closed 2018-01-28 by argot42