freebsd/crochet #208: When Package option is specified, top+work/overlay is not applied

gczuczy, 2017-08-22:
Hello,

I just wanted to rebuild an image with an older config, and I was surprised that the package installation didn't work. As it turned out, the overlay wasn't applied, which enabled it to work.

I've tried removing various options, and when I've removed the package-related options, overlays were applied again.

This used to work, so I guess something broke it in the meantime.

Regards,
-czg

gczuczy, 2017-08-23:
Actually, my problem is different, this might just work fine.

Packages are installed at phase 760.
Overlay is handled at phase 790.

Now, in my setup, I need to add a hosts entry, because the package repo is on a public IP, and from the host system that's not reachable, so I have to tweak it first to make it work. So I'm getting the failures because of this.

Otherwise, it makes sense, because the overlay files are also ought to put in the set up config files, which were installed by the package phase.

I know, my case is a bit extreme, however, I think a small hook, something like custom_prepackage_hook wouldn't hurt, to set up anything that's required within the system. For corner cases.

Any thoughts on it?

kientzle, 2017-08-31:
Does the `_PACKAGE_SITE` argument for `PackageInit` help you here?

Otherwise, you can use `strategy_add 750 my_custom_function` in your config file to do anything you want before the package installation.  The config file is just another shell script, after all.  This has a small risk that future changes to crochet might someday break it, but this is unlikely.

The current ordering of package and overlay installation is intended to address the difference between the build-time environment and the run-time environment.  (The overlay may install networking config that only applies to the environment where the image is ultimately running and can actually break networking at the time the image is being built.)