freebsd/crochet #211: How to start FreeBSD on Orange PC ?

eleutherius, 2017-09-12:
Hello gues, I have problem with starting FreeBSD Current on  [ORANGE PI PC](http://www.orangepi.org/orangepipc/)
I tried to make  an image for OrangePi-Plus2E, but it did not work.
```
U-Boot 2017.01-rc3 (Aug 12 2017 - 22:48:07 +0000) Allwinner Technology
arm-none-eabi-gcc (FreeBSD Ports Collection for armnoneeabi) 6.3.0
GNU ld (GNU Binutils) 2.28
=> boot
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found FreeBSD U-Boot Loader (bin)
reading ubldr.bin
239144 bytes read in 35 ms (6.5 MiB/s)
## Starting application at 0x42000000 ...
Consoles: U-Boot console
Compatible U-Boot API signature found @0x7bf3f6b8

FreeBSD/armv6 U-Boot loader, Revision 1.2
(Mon Sep 11 16:20:55 EEST 2017 alex@FreeBSD)

DRAM: 1024MB
Card did not respond to voltage select!
MMC Device 2 not found
MMC Device 3 not found
Number of U-Boot devices: 1
U-Boot env: loaderdev not set, will probe all devices.
Found U-Boot device: disk
  Probing all disk devices...
  Checking unit=0 slice=<auto> partition=<auto>... good.
Booting from disk0s2a:
/boot/kernel/kernel text=0x74ff3c data=0x71110+0x1d9fb0 syms=[0x4+0x8df30+0x4+0xcc36b]

Hit [Enter] to boot immediately, or any other key for command prompt.
Booting [/boot/kernel/kernel]...
No valid device tree blob found!
```
I tried to choose dtb but it did not lead to success.

```
loader> ls /boot/dtb/
/boot/dtb/
    nanopi-neo.dtb
    orangepi-plus-2e.dtb
    sinovoip-bpi-m3.dtb
    sun4i-a10-cubieboard.dtb
    sun4i-a10-olinuxino-lime.dtb
    sun6i-a31s-sinovoip-bpi-m2.dtb
    sun5i-a13-olinuxino.dtb
    sun5i-r8-chip.dtb
    sun7i-a20-bananapi.dtb
    sun7i-a20-cubieboard2.dtb
    sun7i-a20-olimex-som-evb.dtb
    sun7i-a20-pcduino3.dtb
    sun8i-h2-plus-orangepi-zero.dtb
    sun8i-h3-nanopi-m1.dtb
    ufw.dtb
    sun8i-h3-orangepi-one.dtb
    cubieboard.dtb
    olinuxino-lime.dtb
    bananapim2.dtb
    bananapi.dtb
    cubieboard2.dtb
    olimex-a20-som-evb.dtb
    pcduino3.dtb
    sun8i-h3-nanopi-neo.dtb
    am335x-bone.dtb
    am335x-boneblack.dtb
    am335x-bonegreen.dtb
    beaglebone.dtb
    beaglebone-black.dtb
[... 153 lines cut ...]
         r6 = 0xc4130a80  r7 = 0xc08db4f0
         r8 = 0x7fffffff r10 = 0xc09470c8
bus_set_pass() at bus_set_pass+0x94
         pc = 0xc0296580  lr = 0xc0200094 (mi_startup+0xfc)
         sp = 0xc0c13e70  fp = 0xc0c13e90
         r4 = 0xc09471c4  r5 = 0x00000001
         r6 = 0xc06af29c  r7 = 0x00000000
         r8 = 0xc09471c0  r9 = 0xc06b01c0
mi_startup() at mi_startup+0xfc
         pc = 0xc0200094  lr = 0xc00002c4 (btext+0x144)
         sp = 0xc0c13e98  fp = 0x00000000
         r4 = 0xc00003f8  r5 = 0xc0998000
         r6 = 0x42056500  r7 = 0x00c52078
         r8 = 0xc0b06000  r9 = 0x00000014
        r10 = 0x0000000a
btext() at btext+0x144
         pc = 0xc00002c4  lr = 0xc00002c4 (btext+0x144)
         sp = 0xc0c13e98  fp = 0x00000000
KDB: enter: panic
[ thread pid 0 tid 100000 ]
Stopped at      $d.3:   ldrb    r15, [r15, r15, ror r15]!
db> -
```
Please, help me start FreeBSD on OrangePC. 
Thanks.

arrowd, 2017-09-12:
I had limited success with Orange Pi PC.

First, I had to apply this patch to prevent kernel panic: http://arrowd.name/h3audio.patch
Next, I came up with following dtb: http;//arrowd.name/sun8i-h3-orangepi-pc.dtb

With these I was able to successfully boot into FreeBSD, but unfortunately ethernet driver does not work.

eleutherius, 2017-09-12:
How to insert your custom .dtb file into FreeBSD image ?

arrowd, 2017-09-12:
Create a memory disk from it (`man mdconfig`), mount FreeBSD partition (md0s2a, probably) then copy that dtb into /mnt/boot/dtb.

eleutherius, 2017-09-13:
I was able to install the correct dtb, but how did you install the patch?

arrowd, 2017-09-13:
Well, `cd /usr/src && patch < patchfile`. Then rebuilt the image using crochet. Make sure to remove `crochet/work/._builtkernel_something` to make crochet recompile the kernel.

myfreeweb, 2018-01-21:
Audio patch has landed a while ago: https://reviews.freebsd.org/D8425

How exactly does the ethernet fail?

arrowd, 2018-01-21:
The patch I posted seems to be a different one, it wasn't committed.

Ethernet wasn't working as no `awg` device was showing up. But this got fixed automagically in a funny way - I inserted a cable into an ethernet port first and then powered up the board. After that the `awg0` device appeared during boot and now it works fine.

Well, there are some hangs when bridging `tap` and `awg`, but that's a different story, I think.

myfreeweb, 2018-02-18:
Built `orangepi-pc` U-Boot and dtb, an image for this board… but hit this problem:
https://lists.freebsd.org/pipermail/freebsd-arm/2017-November/017104.html (originally on Plus2E)

So I had to `load -t dtb /boot/dtb/sun8i-h3-nanopi-neo.dtb` :D And that works, but we should be using our fdt…

myfreeweb, 2018-02-18:
hmmm I guess that kernel panic might be audio related, since it happens right after probing pcm.

myfreeweb, 2018-02-18:
Oh @arrowd your patch applies after the committed one. Where did it come from?

myfreeweb, 2018-02-18:
ehhh it doesn't seem to help, just adds the `H3 Audio Codec (analog part)` but the kernel still doesn't boot.

Posted some stuff to bugzilla:

- [226011](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=226011) — fdt
- [226013](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=226013) — u-boot
- [226014](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=226014) — crash

arrowd, 2018-02-19:
> Oh @arrowd your patch applies after the committed one. Where did it come from?

I bugged **jmcneill** on #bsdmips IRC channel, he gave me it. It helped me with the crash, though.

laytonwang, 2018-06-08:
Succeed in compiling Orange Pi One on FreeBSD 12.0-CURRENT #0 r334337(released on Tue May 29 17:12:35 UTC 2018) and set target to armv6 or armv7 by "-b" parameter .
But hung when booting above images either armv6 or armv7 , error message is the following:
Loaded DTB from file 'sun8i-h3-orangepi-one.dtb'
kernel entry at 0x42200100...
kernel args: (null)

Any successful runing on Orange Pi One ? I am appreciated if anyone may let me know the cause of this error or your detailed and correct system environment . BTW , I couldn't get FreeBSD 12.0-CURRENT r333094 as mentioned in "http://occnc.com/info/arm64/NOTES.html" above.

Thanks!

CurtisVillamizar, 2018-06-09:
A board directory for Orange Pi One is now available.  I have two of them and both work fine.  I should update the NOTES.html file.  I have not tried Orange Pi PC (I don't have one or a need for one at the moment).  The Orange Pi PC seems more similar to Orange Pi PC Plus but it would be best to create a new board directory and use the correct dts file.

CurtisVillamizar, 2018-06-09:
Looks like one of the cards that I thought was Orange Pi PC Plus is actually Orange Pi PC but it is running the Orange Pi PC Plus board config.  The marvels of buying on Amazon - wrong card.  BTW you don't have to run a specific build of FreeBSD 12.  So far it has worked on two versions that I pulled.  I mentioned the version just as an FYI.

myfreeweb, 2018-06-09:
Hm, so the dtb has arrived [in the linux 4.15 sync](https://github.com/freebsd/freebsd/commit/d528531ffa71af7f60a5ad0254eed2b20316f341), the u-boot port was merged, no updates on [the audio bug](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=226014) though...

CurtisVillamizar, 2018-06-11:
FYI- I just created a pull request for an Orange Pi PC board directory.  And yes the audio bug is still there.  btw- I updated http://occnc.com/info/arm64/NOTES.html a day or two ago.  It does not reflect the pull request just made so I supposed needs another small update but I'll wait until the pull request is acted on.