freebsd/crochet #216: Running (newly compiled) executables from NFS mount fails with "killed: text file modification"

adriaandegroot, 2017-11-15:
(*This isn't nescessarily a crochet issue, more a FreeBSD-on-Pine64 issue; the FBSD forums don't seem to be the right place, though*)

Running 12-current(-ish) on a Pine64+ 2GB, from the latest RaspBSD image, `FreeBSD 12.0-CURRENT #0 r312006M`, I'm hitting an issue mentioned in #193, and also filed by bdrewery@ in a different repo https://github.com/wraith/wraith/issues/99 .

I'm trying to build ports on the Pine64 (partly to compare speed against building them via QEMU and Poudriere on an amd64 box), and configure keeps crapping out (e.g. in devel/gettext-tools) as soon as it tries to run ./conftest (e.g. when checking for cross-compilation). It's not specific to ports, though, it's got something to do with NFS.

* /usr/ports and /usr/ports/distfiles are NFS mounted from elsewhere
* To save wear on the microSD card, $WRKDIRPREFIX is /usr/ports/tmp, which is writable

The NFS server is an 11-stable box, amd64, with no fancy settings that I know of.

```
    % cd /usr/ports/tmp
    % echo "int main() { return 0 ; }" > t.c
    % cc -o t t.c
    % ./t
    Killed
    Nov 15 14:10:08 pine64 kernel: pid 5507 (sh), uid 0, was killed: text file mod
ification
```
Doing the same from a different amd64 client runnint 10.3 does work: the executable "t" is run and returns 0. 12-CURRENT amd64 also works. If I copy the generated executable from the NFS mount to, say, `/tmp/`, then it runs normally. So (I conclude) it's not a problem with the executable itself, but with the way the runtime linker loads executables, and how that interacts with NFS mounts. But if I copy /bin/ls into the NFS directory, it does work:
```
    # cp /bin/ls ./t
    # ./t
    t       t-amd64 t.c
```

adriaandegroot, 2017-11-15:
file(1) tells me that /bin/ls is version 1 (SYSV), while newly-compiled stuff is version 1 (FreeBSD). Also, /bin/ls is stripped, and newly compiled isn't. And it's the stripped bit that is killing:
```
    # cc -o t t.c
    # ./t
    Killed
    Nov 15 17:31:52 pine64 kernel: pid 20035 (sh), uid 0, was killed: text file modification
    # strip t
    # ./t
```

brd, 2017-11-15:
@adriaandegroot Yeah, that image is quite old.. I am trying to convert/update the build process to work on a new machine.. I hope to have a new image this week.

adriaandegroot, 2017-11-15:
Er, .. I should say "the **unstripped** binary won't run from an NFS mount".

In any case, if you'd prefer a PR (I'm not sure against what, though), just say so; or even "you're a committer, figure it out" (though that's kinda far from my usual stomping grounds).

brd, 2017-11-15:
I think it is just because that image is so far out of date.. but I am not sure..

brd, 2017-11-15:
(I am a committer too, and that is out of my area of expertise too :) We can't know everything..)

adriaandegroot, 2017-11-21:
Closing this, since it's really not a crochet thing. Something to take up on the freebsd-arm mailing list instead (in the mean time, trying to build new images or buildworld on the Pine64 to see if that helps).

closed 2017-11-21 by adriaandegroot