freebsd/crochet #220: RaspberryPi2 board setup.sh sets KERNCONF=GENERIC, but GENERIC does not exist

sonologic, 2017-12-10:
Not sure if i'm doing something wrong, and this used to work before, but on my latest attempt to build an image for the rpi2, I got an error:

```Didn't find /usr/src/freebsd_rpi2b/crochet/src/sys/arm/conf/GENERIC```

And indeed this file does not exist in the releng/11.1 branch:

https://github.com/freebsd/freebsd/tree/releng/11.1/sys/arm/conf

I'm pretty sure that my previous image did build with the RPI2 config, given the output of uname when running that image:

I see the KERNCONF was changed in the board's setup.sh here:

https://github.com/freebsd/crochet/commit/cf14b342e312ba649dffa0ac4921118da8a1d214#diff-8ba6e6ecb3698ee1596ff547ed430cac

What I can't figure out is how this is supposed to work with a non-existing GENERIC kernel config. Of course, I can set KERNCONF=RPI2 in config.sh, but I would expect the default to use that instead of GENERIC.

sonologic, 2017-12-10:
Hmm, that's not the only problem i'm getting it seems. Building now stops with another error:

```cd: stand: No such file or directory```

Is crochet supposed to work with 11.1 at all, or am I just trying some combination that I shouldn't?

snail59, 2018-01-08:
Hello

I had the same problems as you.

1.  KERNCONF
Installing **gsed** corrected this issue.

2. stand
The commit 5222b4da9a7f38daffb84c46c132aa649dd721ae reads "**Reflect base r325834- sys/boot has relocated to stand**". \
But this r325834 commit is only present in  **head**, not in **freebsd 11.1**.
Waiting for a more " adapted" solution, reverting the commit did the trick for me :
` git revert 5222b4da9a7f38daffb84c46c132aa649dd721ae`

brd, 2018-01-16:
Generally we target HEAD because that is what will support the boards the best.

Though handling this properly is probably a good idea for older boards.

closed 2018-07-09 by brd