freebsd/crochet #226: RPI3 boot from USB disk

brianthetall, 2018-01-16:
I'm trying to boot from SD to USB harddisk.

I found this line inside the kernel source, which looks like it would do the trick.
However there is no "RPI3" version of this file.
How does one change the device to be used to load the kernel for a RPI3?

/src/sys/arm/conf/RPI2:
options        ROOTDEVNAME=\"ufs:mmcsd0s2\"

brianthetall, 2018-01-17:
Still not what I need, but I have found that removing the UFS partition from the SD card causes the system to boot to a loader prompt. From there I can specify ufs:/dev/da0s2a; the SSD then boots as root!
....But really need this to happen automatically...

keithmj114, 2018-01-17:
Manually boot into the SSD as root and edit /etc/fstab to point its root directory to /dev/<SSD ID>. Crochet automatically sets root to mount an SD card, changing this should fix the problem.

closed 2018-01-17 by brianthetall