freebsd/crochet #231: freebsd-crochet no longer works with u-boot-rpi2

ivr588, 2018-03-04:
Hi,

It seems firmware files are separated from `sysutils/u-boot-rpi2` port and are now located in `sysutils/rpi-firmware` port as of [ports r454265](https://svnweb.freebsd.org/ports?view=revision&revision=454265), but crochet doesn't take account of this presently and still expects to find files in /usr/local/share/u-boot/u-boot-rpi2

Please fix, many thanks.

cheasy, 2018-04-10:
also true for RPi 3.

brd, 2018-04-11:
Yeah, I am testing fixes.

cheasy, 2018-04-12:
Maybe unrelated, `FreeBSD freebsd12-current.local 12.0-CURRENT FreeBSD 12.0-CURRENT #0 **r331740**: Thu Mar 29 21:24:24 UTC 2018     root@releng3.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64`
does not recompile itself. It dies while compiling clang at
```
--- ASTMatchers/Dynamic/Registry.o ---
c++: error: unable to execute command: Killed
c++: error: clang frontend command failed due to signal (use -v to see invocation)
FreeBSD clang version 6.0.0 (tags/RELEASE_600/final 326565) (based on LLVM 6.0.0)
Target: x86_64-unknown-freebsd12.0
Thread model: posix
InstalledDir: /usr/bin
c++: note: diagnostic msg: PLEASE submit a bug report to https://bugs.freebsd.org/submit/ and include the crash backtrace, preproces
sed source, and associated run script.
c++: note: diagnostic msg:
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
c++: note: diagnostic msg: /tmp/Registry-4d01e3.cpp
c++: note: diagnostic msg: /tmp/Registry-4d01e3.sh
c++: note: diagnostic msg:

********************
*** [ASTMatchers/Dynamic/Registry.o] Error code 254

make[4]: stopped in /usr/src/lib/clang/libclang
1 error
```

brd, 2018-04-13:
Why was it killed?  Check dmesg.

cheasy, 2018-04-15:
@brd : pid 15127 (c++), uid 0, was killed: out of swap space
My bad, should have checked before. Pls close this case.

closed 2018-04-15 by brd