freebsd/crochet #234: u-boot/u-boot-rpi3/armstub8.bin: No such file or directory

cheasy, 2018-04-15:
Building image fails w/above message after world compilation completes.
```
# uname -a
FreeBSD FreeBSD-11.virtual 11.1-STABLE FreeBSD 11.1-STABLE #0 r331742: Thu Mar 29 21:30:37 UTC 2018     root@releng2.nyi.freebsd.org:/usr/obj/usr/src/sys/GENERIC  amd64
# /bin/sh crochet.sh -c config.sh
Starting at Sun Apr 15 16:08:36 UTC 2018
Loading configuration from config.sh
Board: RaspberryPi3
Option: ImageSize 3900mb
....
Mounting all file systems:
Mounting FAT partition 1 at /root/crochet/work/_.mount.boot
Mounting UFS partition 1 at /root/crochet/work/_.mount.freebsd
cp: /usr/local/share/u-boot/u-boot-rpi3/armstub8.bin: No such file or directory
# ls -l /usr/local/share/u-boot/u-boot-rpi3/
total 484
-rw-r--r--  1 root  wheel     123 Apr 15 11:26 README
-rw-r--r--  1 root  wheel  441992 Apr 15 11:26 u-boot.bin
```

brd, 2018-04-15:
Update your checkout of crochet, this has been fixed.

closed 2018-04-15 by brd

cheasy, 2018-04-15:
Checked, builds fine. Thanks a lot.