freebsd/crochet #238: Heads UP: please update the status

PetruGarstea, 2018-04-19:
Greetings,

I haven't touched crochet for over 6 months, I wanted to build a new image and during debugging I found that

1. Crochet now supports only head, what about 11 release branch ?
2. Crochet cannot create the image because there is an issue with u-boot-rpi2 package

Please update the status of the project or where can i find future plans, changelogs and other news.

Cheers,
Petru

brd, 2018-04-19:
1. Crochet has only ever targeted HEAD.  If it happens to work on 11 that is fine, but not something we are that concerned with.  This is for multiple reasons, but really it boils down that HEAD supports more devices better.  Many large changes won't be merged back to 11.
2. Yeah, I need to work on testing this.

Mostly my time is focused on the pkgbase project since that represents the best way to update ARM devices and is really important for FreeBSD on ARM.

tickerguy, 2018-04-19:
I have a "working" Crochet for 11-Stable that will build on the RPI2, if you need it.. .I could make it available.  It's a bit older, but functional and is what I use for the RPI2 series.

I haven't built on it for the last couple of months but it certainly should still work...

PetruGarstea, 2018-04-19:
Hi,

    I can download 11.1 from from freebsd repo, thx.

I have a list of packages and configs that I baked into my image with crochet. If i go with clean install then it takes too long time to install packages.

I believe there is a migration from u-boot-rpiX to rpi-firmware package, but I cannot say for sure, I didn't find any official information.

pjbloknl, 2018-04-20:
Just to be sure. Is necessary to run crochet on -current or can it run on -stable? I do run it on -stable with a -current source and so far that is ok.

Also I have used a union mount to merge rpi-firmware over u-boot-rpi2 which works fine. rpi1 needs different approach

brd, 2018-04-20:
Nope, building on -STABLE is fine.  Assuming you are only one major version number behind as FreeBSD only guarantees that.

PetruGarstea, 2018-04-21:
I have built a new image from head branch with a copy of rpi-firmware to u-boot-rpi2 folder under /usr/local/share.

When booting uboot doesnt load the kernel it stops with boottp.

thx

DenisPolygalov, 2018-06-13:
FYI: Building on 11.1-RELEASE and earlier fails due to absence of the whole /usr/src/stand tree in the source code. So crocket is not useful on the **latest** RELEASE version of FreeBSD which is quite sad...

PetruGarstea, 2018-06-30:
11.2 it works, you will have to create a symlink to dst file in src tree

1. mkdir /usr/src/sys/dts/
2. ln -s /usr/src/sys/boot/fdt/dts/arm/rpi2.dts /usr/src/sys/dts/rpi2.dts

DenisPolygalov, 2018-07-03:
@PetruGarstea  why 'arm' folder is missing? Not needed? Maybe:

`# mkdir -p /usr/src/sys/dts/arm`
`# ln -s /usr/src/sys/boot/fdt/dts/arm/rpi2.dts /usr/src/sys/dts/arm/rpi2.dts`

PetruGarstea, 2018-07-07:
@DenisPolygalov in 11 and 12 versions the path is different and crochet is targeting only with the head, the above small changes allows to build 11.2 release with crochet