freebsd/crochet #269: Installworld failed:ERROR: Required ntpd user is missing

zqb-all, 2018-11-09:
I try to build img for pine64. then fail when installworld.

```bash
$ cat /usr/crochet/work/_.installworld.aarch64.log
make[1]: "/usr/crochet/work/obj/usr/src/arm64.aarch64/toolchain-metadata.mk" line 1: Using cached toolchain metadata from build at freebsd on Thu Nov  8 11:24:53 UTC 2018
ERROR: Required ntpd user is missing, see /usr/src/UPDATING.
*** Error code 1

Stop.
make[1]: stopped in /usr/src
*** Error code 1

Stop.
make: stopped in /usr/src
```

zqb-all, 2018-11-12:
This issue occur when build in "freebsd 11.02".
Finally I use "freebsd 12.0-BETA3 amd64" in my virtualbox to build and everything is ok.

kozross, 2018-11-12:
@zqb-all I'm working in a qemu image, and FreeBSDs later than 11.2 won't even boot.

closed 2018-11-21 by zqb-all

tldrm, 2018-12-21:
Getting this error now.
Raspberry Pi, FreeBSD 12.0-RELEASE.  The ntpd user and group are present in the building host.  src is borrowed from my poudriere jail created just today (poudriere jail -c -j FreeBSD:12:amd64 -v 12.0-RELEASE -a amd64).  My crochet is also from today (master, zip).

reezer, 2019-01-03:
I experience the same problem as @tldrm Maybe this bug should be reopened.

przzzp, 2019-01-08:
After adding the user ntdp, you must update the password database:
# pwd_mkdb /etc/master.passwd
Result:
# id -u ntpd
123

iseus, 2019-01-08:
This is still not a crochet issue. In the other hand, use pw to create groups and users, it will do the background work for you.

przzzp, 2019-01-08:
The fact is that when upgrading from version 11 to version 12, the system adds entries to the /etc/passwd and /etc/group files, but does not update the database.

iseus, 2019-01-08:
> The fact is that when upgrading from version 11 to version 12, the system adds entries to the /etc/passwd and /etc/group files, but does not update the database.

Thanks, I didn't noticed this yet. I'm still using 11.1 on my build host.

Ofloo, 2019-01-28:
/usr/src # make -j9 installworld
--- installworld ---
make[1]: "/usr/obj/usr/src/amd64.amd64/toolchain-metadata.mk" line 1: Using cached toolchain metadata from build at azkaban.ofloo.net on Sun Jan 27 13:17:21 CET 2019
--- __installcheck_UGID ---
--- __installcheck_sh_check ---
--- __installcheck_UGID ---
ERROR: Required ntpd user is missing, see /usr/src/UPDATING.
*** [__installcheck_UGID] Error code 1

make[1]: stopped in /usr/src
1 error

make[1]: stopped in /usr/src
*** [installworld] Error code 2

make: stopped in /usr/src
1 error

make: stopped in /usr/src