freebsd/crochet #293: Can crochet run from Linux?

ecorm, 2023-06-04:
I'm interested in generating custom FreeBSD images for a BeagleBone Black device, but my desktop development environment is a Linux-based distro. Can crochet run from Linux, or do I need to set up a FreeBSD build PC for this purpose?

I'm already familiar with Buildroot for generating embedded Linux images, but I'm looking into migrating to FreeBSD for a commercial project to avoid legal/licensing issues.

brd, 2023-06-05:
Hello!  There is some work to build FreeBSD on linux, but I doubt crochet knows how to take advantage of it.  Maybe you could use a FreeBSD VM to do it?

These days I would probably look at [poudriere](https://www.freshports.org/ports-mgmt/poudriere-devel/) image support, it can spit out a raw filesystem image that you can then add the appropriate uboot, etc to using mkimg(1) etc.

ecorm, 2023-06-05:
Thanks for the info, brd!

closed 2023-06-05 by ecorm

ecorm, 2023-06-05:
@brd, if I understand correctly, both crochet and pourdriere are for cross-compiling the basic OS only, and if I want to add libraries or userland applications, it must be done on the device via the command line.

I need to be able to generate a *complete* image, including libraries, 3rd party apps, and our in-house app that can be deployed all at once on the device. Buildroot allows me to do that (with Linux as the kernel), but I don't see an equivalent way to do this for FreeBSD.

I hope I am mistaken about these limitations regarding crochet/poudriere.

ecorm, 2023-06-05:
I found the poudriere discussion forum on GitHub and made an inquiry there: https://github.com/freebsd/poudriere/discussions/1056